Ysurac/openmptcprouter #2083: Impossibile ping or resolve ping or addres via default route

tanganellilore, 2021-09-19:
## Expected Behavior

ping or resolve everything from all intarfaces
## Current Behavior

Impossibile to ping internet IP or resolve any internet address via default route (Tun0 callwd omrvpn) 
## Specifications

  - OpenMPTCProuter version: 0.58
  - OpenMPTCProuter VPS version:  0.1026
  - OpenMPTCProuter VPS provider: Other
  - OpenMPTCProuter platform: RPI4 

Other info:
I'm not able to ping or resolve anything if I'm not specify the Interface. 
I'm able to ping vpn tunnel from vps and from router. 
I'm able to navigate if i put my default router as DNS. 
Configuration performed via wizard

Command `route`:
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.255.255.1    0.0.0.0         UG    0      0        0 tun0
default         192.168.1.1     0.0.0.0         UG    3      0        0 wan1
default         192.168.1.1     0.0.0.0         UG    6      0        0 eth0
default         10.255.255.1    0.0.0.0         UG    1200   0        0 tun0
10.255.255.1    *               255.255.255.255 UH    0      0        0 tun0
10.255.255.2    *               255.255.255.255 UH    1200   0        0 tun0
127.0.0.0       *               255.0.0.0       U     5      0        0 lo
185.25.205.222  192.168.1.1     255.255.255.255 UGH   1      0        0 wan1
192.168.1.0     *               255.255.255.0   U     3      0        0 wan1
192.168.1.0     *               255.255.255.0   U     6      0        0 eth0

Seems that something block my traffic:
If i disable temporary the omrvpn everything work but I recive other error relate vpn down (obviously) 

Any suggestion? 
Many thanks for tour help

Ysurac, 2021-09-19:
Everything is green in status page ?
What is the exact version ? if 0.58, update to latest.
What is the VPS provider ? Did you reboot VPS and all is working correctly on VPS ?

tanganellilore, 2021-09-19:
Sorry, I missing minor release.
Rp4 have 0.58.5

Related omr status, below the picture:
[image]

If i remove vpn:
[image]

If I don't remove vpn first time i recive warning:
Big time differance beetwen...
Not possibile retrieve public ip
Not possibile retrieve IP from shadosocks.

Ysurac, 2021-09-19:
So did you check time on both end ?
If you don't have DNS errors, then it's working on router side, you can test via SSH.

tanganellilore, 2021-09-19:
After sync via browser yes.
If i try to ping, dig, nslookup via ssh on router, but no response without Interface specified.

Ysurac, 2021-09-19:
can you ping 10.255.255.1 ? (if glorytun VPN is used)

tanganellilore, 2021-09-19:
Yes:

[image]

Ysurac, 2021-09-19:
Did you change DNS configuration ?
Did you try to disable DNSSEC ?

tanganellilore, 2021-09-19:
For dns, I think no.

Wan1 is setted via static ip
Eth0 is setted as static ip and Dhcp disable.

Related dnssec, which setting i need to disable?

Ysurac, 2021-09-19:
In Network->DHCP&DNS disable DNSSEC in advanced setting tab and in services->DNS recursive, uncheck "Enable DNSSEC"

tanganellilore, 2021-09-19:
No change, same issue.
[image]

I need to reboot?

Ysurac, 2021-09-19:
If you can ping VPS side of VPN, default route is VPN IP, then issue is on VPS side that doesn't seems to route correctly.
Check that shorewall is running.

tanganellilore, 2021-09-19:
I agree with you and there is shorewall with error on vps side. 

I try to re-enable service but vm go down, I need to check in detail the rules if something go wrong. 

I will update you in netx hours/days.

tanganellilore, 2021-09-20:
@Ysurac Many thanks for suggestion!!!

After fix shorewall rules on vps side everything works well.
I need to re-enable DNS and other setting removed?

Thanks

tanganellilore, 2021-09-23:
everything works after fix shorewall rule with default one.

Thanks

closed 2021-09-23 by tanganellilore