Ysurac/openmptcprouter #2085: Only Master Connection is used

eothein, 2021-09-20:
I have the following setup:

[image]

The Mikrotik routers are doing some traffic shaping. I installed another server in the network with an InfluxDB, Telegram and a Grafana server to do some monitoring. 

It seems that my setup is only using the master connection when downloading a file: `wget http://ipv4.download.thinkbroadband.com/1GB.zip` while I would think all my wan links would be used. 

Am I doing something wrong here?

Kind regards
Jens Buysse

eothein, 2021-09-21:
Dear Ysurac

For a general question there was no template available, or I must have missed it? 
With regards to the Kernel remark, I just used a Ubuntu LTE image from Google Console - which is supported by the documentation. Am I missing something?

Kind regards
Jens Buysse

Ysurac, 2021-09-21:
True, as indicated in issue part, question are for part not related directly to OpenMPTCProuter, wong section ;)
Debian is the recommended OS for VPS part. Check that you can use a custom kernel on the VPS (I don't remember if there is a special setting or not).

eothein, 2021-09-22:
Ok I'll fire up a Debian VPS and install OpenMPTCPRouter's VPS on it: I'll keep you posted.

eothein, 2021-09-22:
Hey @Ysurac , installing and configuring the Debian VPS seems to solve the issue. Alhtough, the documentation states there is support for  Ubuntu Server 20.04 x86_64. Is this then still correct? 

What do you think the difference between those two packages could be, why did it go wrong? This would be very interesting for us to identify.

closed 2021-09-27 by eothein