Ysurac/openmptcprouter #2086: VPS on linode throughput limited after install script

shamen123, 2021-09-20:
## Expected Behavior
The VPS install script should not limit VPS network throughput. However, eth0 throughput is significantly limited after installing. The connection from  Tested on three different linode instances and regions. Thoughput is high before VPS script install, and broken afterwards. 

## Current Behavior
1. Provision Linode VPS (I used 4Gb dedicated debian 10). 
2. Install speed test and validate throughput on VPS
```
Speedtest by Ookla

     Server: Structured Communications - London (id = 11445)
        ISP: Linode
    Latency:     0.84 ms   (0.03 ms jitter)
   Download:  1564.56 Mbps (data used: 2.3 GB)                               
     Upload:  4016.05 Mbps [=========/          ] 48%  ^C
```

3. Run VPS installer 
`wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh
`
4. Reboot machine
5. re-run speed test 
`
   Speedtest by Ookla

     Server: Structured Communications - London (id = 11445)
        ISP: Linode
    Latency:     0.85 ms   (0.04 ms jitter)
   Download:   117.96 Mbps (data used: 93.5 MB)                               
     Upload:    56.58 Mbps [===|                ] 16%`

The throughput to/from VPS is significantly reduced simply by installing the VPS scripts and rebooting. 

This translates to poor throughput to the router:

OMR iperf
Connecting to host 176.58.109.70, port 65400
Reverse mode, remote host 176.58.109.70 is sending
[  5] local 212.69.62.95 port 18941 connected to 176.58.109.70 port 65400
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  88.8 KBytes   727 Kbits/sec                  
[  5]   1.00-2.00   sec   300 KBytes  2.45 Mbits/sec                  
[  5]   2.00-3.00   sec   419 KBytes  3.43 Mbits/sec                  
[  5]   3.00-4.00   sec   528 KBytes  4.33 Mbits/sec                  
[  5]   4.00-5.00   sec   925 KBytes  7.58 Mbits/sec                  
[  5]   5.00-6.00   sec  1.20 MBytes  10.1 Mbits/sec                  
[  5]   6.00-7.00   sec  1.96 MBytes  16.4 Mbits/sec                  
[  5]   7.00-8.00   sec  3.00 MBytes  25.1 Mbits/sec                  
[  5]   8.00-9.00   sec  4.27 MBytes  35.8 Mbits/sec                  
[  5]   9.00-10.00  sec  6.07 MBytes  50.9 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.11  sec  22.5 MBytes  18.7 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  18.7 MBytes  15.7 Mbits/sec                  receiver

iperf Done.
root@OpenMPTCProuter:~# omr-iperf vps 
Connecting to host 176.58.109.70, port 65400
[  5] local 212.69.62.95 port 19407 connected to 176.58.109.70 port 65400
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  16.1 MBytes   135 Mbits/sec    0   14.1 KBytes       
[  5]   1.00-2.00   sec  29.7 MBytes   249 Mbits/sec    0   14.1 KBytes       
[  5]   2.00-3.00   sec  26.2 MBytes   220 Mbits/sec    0   14.1 KBytes       
[  5]   3.00-4.00   sec  21.2 MBytes   178 Mbits/sec    0   14.1 KBytes       
[  5]   4.00-5.00   sec  23.8 MBytes   199 Mbits/sec    0   14.1 KBytes       
[  5]   5.00-6.00   sec  22.5 MBytes   189 Mbits/sec    0   14.1 KBytes       
[  5]   6.00-7.00   sec  22.5 MBytes   189 Mbits/sec    0   14.1 KBytes       
[  5]   7.00-8.00   sec  26.2 MBytes   220 Mbits/sec    0   14.1 KBytes       
[  5]   8.00-9.00   sec  23.8 MBytes   199 Mbits/sec    0   14.1 KBytes       
[  5]   9.00-10.00  sec  25.0 MBytes   210 Mbits/sec    0   14.1 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   237 MBytes   199 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   231 MBytes   194 Mbits/sec                  receiver

Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 9536M    4  381M    0     0  75.8M      0  0:02:05  0:00:05  0:02:00 76.3M

## Specifications

  - OpenMPTCProuter version: 0.5.85
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: linode
  - OpenMPTCProuter platform: x86_64
 - ISP A : 1000down/100 up
 - ISP B: 1000 down/100 up

shamen123, 2021-09-20:
I have tried the same using the ubuntu 20.04 template and script - I see the same issue.. 

As you see, speedtest shows Gigabit speeds.. then install VPS script and reboot and throughput drops to 100mbps. 

There is fine throughput from VPS to internet before installation, very poor throughput after. 

I have copied the entire log into a pastebin so you can see what i mean (the instance is already destroyed so there is no issue with the keys being in here) 

[https://pastebin.com/biBZU8m4](url)

Ysurac, 2021-09-21:
The last check from the router give a good result: it's 76.3Mo/s, so 76.3*8=610.4Mb/s, not so bad (at least then previous speedtest).
You can try to run ```omr-test-speed``` from the VPS.

shamen123, 2021-09-21:
@Ysurac that is still significantly limited to what it used to be, and the connection setup time is significantly longer (overhead). Has the VPS recipe or Kernel changed?

Lets look at the primary situation

Fresh install speed test:

```
   Download:  2000.21 Mbps (data used: 2.9 GB)                               
     Upload:  3995.10 Mbps (data used: 6.4 GB)                               
Packet Loss:     0.0%
 Result URL: https://www.speedtest.net/result/c/7e02e988-ae10-49ed-957b-7702c37249c7
```

Then I install VPS script and reboot

```
    Latency:     0.83 ms   (0.55 ms jitter)
   Download:   112.04 Mbps (data used: 77.7 MB)                               
     Upload:    56.82 Mbps (data used: 26.3 MB)                               
Packet Loss:     0.0%
 Result URL: https://www.speedtest.net/result/c/da608d8f-4b9b-4e3f-871e-4c7f4c097b2d
```

Why does the speed reduce so much just by installing the script?

Ysurac, 2021-09-21:
There is changes on each new script release.
Try to use omr-test-speed, it's possible that there is issue with speedtest and MPTCP.

shamen123, 2021-09-21:
@Ysurac I do not believe this is specific to MPTCP. I booted VPS (after install script) with default kernel 4.19.0 and the speedtest issue was still present. 

I know I am probably the only person mad enough to be trying to use this for bonding FTTP for multi gigabit speed but, why not? lol. The reason I am looking at to what the VPS can do on a single download connection is that (from the network behind the omr, some connections have problems, others do not. Its hard to track down. I can run fast.com in a browser and get 1.5Gpbs. But then I can run iperf3 (to a 2nd server in linode with 4gbps) and I see each connection takes 500ms overhead and then reaches only a few hundred mbps.

So:

- omr-iperf vps -R shows full combined line speed. 1.7Gbps down/210mbps up. Perfect. 
- But some connections are doing only a few mbps - for example NNTP 
- Some connections have significant TCP handshake latency when opening (for example iperf3 from inside network to another linode machine)

So logically this is a little more than just the mptcp, as the omr-iperf works perfectly at full speed. Connections going down v2ray (also tried shadowsocks) are either working mostly well, or working not very well at all. So I focus first on "can the VPS communicate with the remote host at line speed?" - and the answer was "no, not with the openmptcprouter scripts installed" 

Like I said, i tried with 4.19 kernel to rule out MPTCP issue. And the problem is still there. This means its got to be something to do with the way packets are moving. So I tried to look at sysctl settings: 

So, with MPTCP kernel: 
5.4.100-mptcp + /etc/sysctl.d/90-shadowsocks.conf == speedtest runs ~100mbps
5.4.100-mptcp without /etc/sysctl.d/90-shadowsocks.conf == speedtest runs 5Gbps 

Without MPTCP kernel
4.19.0-17 + /etc/sysctl.d/90-shadowsocks.conf == speedtest runs ~100mbps
4.19.0-17 without /etc/sysctl.d/90-shadowsocks.conf == speedtest runs 5Gbps 

So the problem is not MPTCP - but something in the sysctl settings. 

As it turns out, it is ECN 

```
grep ecn /etc/sysctl.d/90-shadowsocks.conf 
net.ipv4.tcp_ecn=1
root@localhost:~# ./speedtest -s 35057

   Speedtest by Ookla

     Server: Clouvider Ltd - London (id = 35057)
        ISP: Linode
    Latency:     0.81 ms   (0.02 ms jitter)
   Download:    57.53 Mbps [=/                  ]  7%  ^C
root@localhost:~# sysctl -w net.ipv4.tcp_ecn=2
net.ipv4.tcp_ecn = 2
root@localhost:~# ./speedtest -s 35057

   Speedtest by Ookla

     Server: Clouvider Ltd - London (id = 35057)
        ISP: Linode
    Latency:     0.78 ms   (0.03 ms jitter)
   Download:  4876.44 Mbps [=|                  ]  9%  ^C
```

So to make sure this is the same from the home box, behind OMR, i test like this

Test Box (2.5gpbs) -> OMR (2x1Gbps) -> VPS (4Gbps) -> internet

on VPS
```sysctl -w net.ipv4.tcp_ecn=1
net.ipv4.tcp_ecn = 1
```
on TestBox
```testbox:~$ ./speedtest -s 11445

   Speedtest by Ookla

     Server: Structured Communications - London (id = 11445)
        ISP: Linode
    Latency:     9.03 ms   (0.10 ms jitter)
   Download:   116.38 Mbps [============|       ] 63%```

On VPS 
```sysctl -w net.ipv4.tcp_ecn=2
net.ipv4.tcp_ecn = 2
```
On TestBox
```testbox:~$ ./speedtest -s 11445

   Speedtest by Ookla

     Server: Structured Communications - London (id = 11445)
        ISP: Linode
    Latency:     9.07 ms   (0.32 ms jitter)
   Download:  1478.36 Mbps [==-              ```

So it would appear that ECN in 90-shadowsocks.conf is playing a problematic part when set to value 1 with this solution in use on linode. I assume this means linode has some incompatible routing kit somewhere and requires ECN to be set to compatibility mode 2.

Ysurac, 2021-09-21:
True, something on linode seems to doesn't support ECN.
Good find, I will set it to 2, this shouldn't really make any problem.

shamen123, 2021-09-21:
awesome, thanks. 

I agree - setting to 2 should probably be the default as it is not known if there is any equipment incompatible with ECN in the pathways of all of the different providers and end user equipment. 

Good work on this btw - im going to be doing tuning to see just how fast i can get this ;)

dooley1981, 2021-09-26:
How do you set the ECN to 2? I seem to be having the same issues.

Ysurac, 2021-09-26:
As indicated:
```sysctl -w net.ipv4.tcp_ecn=2```

dooley1981, 2021-09-27:
Okay thanks

dooley1981, 2021-09-27:
How do I get to the screen to see if this has been changed. I flashed the new script over my current one.

Ysurac, 2021-09-27:
```sysctl net.ipv4.tcp_ecn```

dooley1981, 2021-09-27:
I'm sorry but I don't know where to look to see that or enter that.

Ysurac, 2021-09-27:
Via SSH on the VPS, you can enter this command. If "2" then change is done, if "1" no changes.

dooley1981, 2021-09-27:
Okay I understand. Thanks

closed 2021-12-28 by shamen123