Ysurac/openmptcprouter #2088: Patch request to fix ipheth urb errors

darthclide, 2021-09-22:
https://github.com/openwrt/openwrt/pull/4084
I am unsure how long they will take until they merge this patch. And I have no idea what version they will add it to. But I think if you got this implemented as soon as possible, it will help anyone out there having Apple tethering issues.

Ysurac, 2021-09-23:
This will be added in next snapshot and in next release when ready.

darthclide, 2021-09-23:
Awesome, let's hope this ends usb tethering problems with Apple products.

darthclide, 2021-09-23:
Do you think this patch will also resolve these kind of errors?
__mptcp_init4_subsockets: token 0xa4c00a8c bind() to 172.20.10.3 index 666 failed, error -99

__mptcp_init4_subsockets errors have only ever shown up alongside ipheth errors.

Ysurac, 2021-09-23:
Possible, this error can happen when an ip doesn't exist anymore, can be when interface doesn't answer.

darthclide, 2021-09-24:
Hmmm, in the meantime, do you have any ideas on how I could get this interface to restart only when these kind of errors occur?

closed 2021-12-28 by github-actions[bot]

darthclide, 2022-03-15:
As mentioned on your irc, the beta did not stop the gateway from going down. So the tetherme developer's guess was wrong about packet size causing the drop. But what I find interesting is that tonight the iPad lost the wan address, yet still maintained a connection to the ipad (see attached photo)

I immediately looked at the ipad and it showed full 5 bars of 4G LTE. And battery was at 100%. I do not know if it is related, but I figured I would report it here for the overall iPad USB ethernet problems.
[image]