Ysurac/openmptcprouter #2090: Vmware Network unreachable

bagou4502, 2021-09-25:
Hey i have a problem with last version of openmptcprouter with vmware workstation 16.
I can't find any way to provide internet connection to the vm
When i run ping (ip) i got : Network unreachable

Ysurac, 2021-09-25:
You need to create interfaces you need or set interface in vmware as promiscuous.

bagou4502, 2021-09-25:
[image]
Actualy i have that in vmware workstation setting.
I have same problem with x86_64 img and virtualbox img.
My box use 192.168.0.1 to 192.168.0.254 can this be a problem?

But how i can create interface?

bagou4502, 2021-09-25:
Update
Actualy openmptcprouter work but i cant connect openmptcprouter to my internet box
Here IS m'y setup
[image]
I have this config 
[image]

bagou4502, 2021-09-25:
Update :

ALL is good just need to see some things

First i have 50% of my vdsl how i can reduce this loss
Second some website like openmptcprouter.com doesn t work i think is a dns server problem but how i can resolve that

closed 2021-12-30 by github-actions[bot]