Ysurac/openmptcprouter #2091: internet speed and ssl problem

bagou4502, 2021-09-25:
Hey with last version of openmptcprouter with x86 img and a ovh fr vps i have 2 big problem
To begin with, I have a huge loss of connection and I have to reload twice each time the sites that I have never visited
Then I have ssl error when requesting http like
cURL error 35: OpenSSL SSL_connect: SSL_ERROR_SYSCAL

bagou4502, 2021-09-25:
and i have this in vps logs

Sep 25 22:53:26 vps-947927b1 ss-server[1589]: getpeername: Transport endpoint is not connected
Sep 25 22:53:26 vps-947927b1 ss-server[1589]: getpeername: Transport endpoint is not connected
Sep 25 22:53:27 vps-947927b1 ss-server[1589]: getpeername: Transport endpoint is not connected

bagou4502, 2021-09-25:
and i have that 

[image]

"The vpn do not respond"
"The ipv6 tunnel do not respond"

Ysurac, 2021-09-26:
Please use the template, I need release, platform,...

bagou4502, 2021-09-26:
> Please use the template, I need release, platform,...
"Hey with last version of openmptcprouter with x86 img "

bagou4502, 2021-09-26:
and same problem with local website on 192.168.x.xx sometime the website take 5min to be load

bagou4502, 2021-09-28:
some news ?

Ysurac, 2021-09-28:
Again, please edit the issue and use the full template. This really help to solve an issue and I don't have to ask always same questions like the VPS used,...
Always add a way to reproduce the issue.
Check also system log to find why VPN is not able to connect.
For speed, describe how you test it. Did you use ```omr-test-speed``` ans ```omr-test-speed wan1```.
You use MacVLAN ? only one ethernet physical interface ?

bagou4502, 2021-09-30:
its a default configuration no change about that
For vpn issue is resolved

For reproduce that just setup a openmttcprouter with a freebox delta and 1 windows pc

closed 2022-01-03 by github-actions[bot]