Ysurac/openmptcprouter #2094: How to enable multipath UDP

eothein, 2021-09-27:
## Expected Behavior
I open a UDP connection I.e., `iperf3 -c 10.10.1.10 -t 120 -u -b 100M` and I see that all the WAN links are being used. 
GloryTun UDP is enabled in the configuration (VPN -> GloryTun UDP -> Enabled (yes)

## Current Behavior
Only one of the WAN connections is used. 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS provider: Google
  - OpenMPTCProuter platform: Intel(R) Core(TM) i7-10710U CPU @ 1.10GHz

Ysurac, 2021-09-27:
You can check via SSH on the router the result of ```glorytun-udp path```

highfreq, 2021-09-28:
Just a quick question, can glorytun udp and tcp be active at the same time or one excludes the other?

Ysurac, 2021-09-28:
Only one VPN can be used and there is no usage to use multiples VPN between router and VPS.

highfreq, 2021-09-28:
Ok, thanks

eothein, 2021-09-30:
Dear @Ysurac 

Issuing that command provide the following output: 
```
root@OpenMPTCProuter:~# glorytun-udp path
no active tunnel
```

Although my test do show that is traffic split over the different links now..? 

Moreover: with the command frop @highfreq , does this mean the setup only supports either TCP or either UDP? Not both protocols at the same time? 

Kind regards
Jens Buysse

Ysurac, 2021-09-30:
@eothein Glorytun UDP doesn't seems to be working, did you enabled it via the wizard or via the VPN menu ? VPN need to be enabled via the wizard.
Glorytun UDP and Glorytun TCP are 2 differents VPN (one using TCP, the other using own aggregation system based on UDP), 
both can transmit TCP and UDP (and ICMP,...) on them.

eothein, 2021-10-13:
So how do I enable GloryTUN UDP or Glorytun TCP in the wizard. It is really unclear how to configure this?

Ysurac, 2021-10-13:
Using the "Show advanced settings" checkbox in the wizard, change the "Default VPN".

closed 2022-01-17 by github-actions[bot]

eothein, 2022-02-01:
Ok so can I get this to work:

- I changed in the Advanced Settings "Default VPN" GlorytunUDP 
- I created a new key using `glorytun keygen` to create a key
- I added the key in the LUCI dashboard

But this does not seem to work. 
Is there a "howto" on how to configure this?

eothein, 2022-02-01:
@Ysurac could you please have a look a this still?

Ysurac, 2022-02-01:
You don't need to create a new key or something like that, only set VPN in the wizard you can try any VPN.

eothein, 2022-02-01:
As I have already changed the key in the config, which key do I need to find (on the server?) Is the same key for the general VPS?

How can I validate that GlorytunUDP is working on the VPS?

eothein, 2022-02-01:
When I set the Default VPN to GlorytunUDP, I get VPN Tunnel down