Ysurac/openmptcprouter #2097: Excessive NAT connections, RAM leak?

gsxryan, 2021-09-30:
## Expected Behavior
There shouldn't be so many NAT connections persisting.  Seems to correlate with a memory leak / excessive RAM allocation.

## Current Behavior
Memory slowly climbing.
[image]

Slow climb correlates with excessive NAT connections in ISP router table.  Hitting the ISP limit may be decreasing performance or causing other issues.
[image]

After rebooting:
[image]

## Possible Solution
Workaround or bugfix for excessively created NAT connections.  They persist and have a very long TTL.  I attempted to reduce TTL in a few places in OMR, but no luck yet to help the issue.

## Steps to Reproduce the Problem

  1.  Wait for high uptime (2-3 days)
  2. Observe High RAM allocated
  3. Monitor ISP router for NAT connections in table.

## Context (Environment)
The performance of OMR's administration panel gets slower and slower as the RAM builds, and the NAT connections increase.  To make a change it takes 10x longer after high uptime, than after a fresh reboot.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS provider: linode
  - OpenMPTCProuter platform: x64, proxmox, ubuntu
  - Country: USA

Ysurac, 2021-09-30:
UPnP is disabled by default, services are not running.
Where do you get NAT table from ?
You can change TTL in System->OpenMPTCProuter, "Advanced settings" tab.

gsxryan, 2021-11-14:
> For the ss-redir problem, do you use P2P or have something that try to contact an IP that is not reachable ?

My UDM pro pings google DNS constantly, but can't find many entries in the NAT table for it.  I'll switch to the local address anyway

gsxryan, 2022-02-12:
This is still an issue.  I'm currently bypassing all my IOT devices outside of MPTCP router to keep the NAT connections low.  I will need some more time to narrow down which device is causing this excess.

closed 2022-05-19 by github-actions[bot]