Ysurac/openmptcprouter #2098: Windows server 2019 no connection

bagou4502, 2021-09-30:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS version: not activated for this test
  - OpenMPTCProuter VPS provider: not activated for this test
  - OpenMPTCProuter platform: x86

Hey my openmptcprouter work but my windows server 2019 with a ad server can t connect to the openmptcprouter but network configuration are same as my pc (just local ip changed) do you know why?

Adorfer, 2021-10-01:
"can't connect"? Please be a bit more specific. What is the actual error message? By which program/setup? While which operation?
(i do run several WS inc. DC on an OMR site without problems so far)

bagou4502, 2021-10-02:
Just no connection with a warning logo on windows ethernet logo

closed 2022-01-05 by github-actions[bot]