Ysurac/openmptcprouter #2099: Package update/install certificate error

Ysurac, 2021-10-01:
## Expected Behavior

No error in package update/install

## Current Behavior

Downloading https://download.openmptcprouter.com/release/v0.58.2/rpi4/targets/bcm27xx/bcm2711/packages/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.2/rpi4/targets/bcm27xx/bcm2711/packages/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.58.2/rpi4/packages/aarch64_cortex-a72/base/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.2/rpi4/packages/aarch64_cortex-a72/base/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.58.2/rpi4/packages/aarch64_cortex-a72/luci/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.2/rpi4/packages/aarch64_cortex-a72/luci/Packages.gz

Downloading https://download.openmptcprouter.com/release/v0.58.2/rpi4/packages/aarch64_cortex-a72/openmptcprouter/Packages.gz
*** Failed to download the package list from https://download.openmptcprouter.com/release/v0.58.2/rpi4/packages/aarch64_cortex-a72/openmptcprouter/Packages.gz

## Possible Solution

Via SSH, use: 
```opkg --no-check-certificate update``` and ```opkg --no-check-certificate install <package>````

lordmelathar, 2021-11-24:
Other fix for me : 

opkg install wget-ssl

dur3x, 2021-12-08:
> Other fix for me :
> 
> opkg install wget-ssl

Thanks, this quick solution fixed my problem too ;-)

closed 2022-03-13 by github-actions[bot]