Ysurac/openmptcprouter #2103: Prime video problem

bagou4502, 2021-10-02:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS version: 0.1026.5.4.100-mptcp
  - OpenMPTCProuter VPS provider: OVHCloud
  - OpenMPTCProuter platform: x86

Hey i have a problem with amazon prime video when i go on primevideo i have that 
[image]

I have that in omr bypass section
[image]

But not work

bagou4502, 2021-10-02:
on freebox its work but alexa not work and on pc not work

Ysurac, 2021-10-02:
OMR-ByPass is made for that. VPS IP are often banned by Netflix, Amazon Video,...
You need to use a VPS that doesn't have IP banned (often small VPS provider), or use OMR-ByPass to use a dedicated connection but you will not have aggregated connection.

bagou4502, 2021-10-05:
do you know french vps provider not banned?

bagou4502, 2021-10-05:
and i see vpn i don t see ip banned men

closed 2022-01-12 by github-actions[bot]