Ysurac/openmptcprouter #2105: Can't install script for debian 10

dooley1981, 2021-10-03:
I'm trying to install the script for debian 10 but i'm getting the following error...

ERROR: The certificate of ‘www.openmptcprouter.com’ is not trusted.
ERROR: The certificate of ‘www.openmptcprouter.com’ has expired.

Ysurac, 2021-10-03:
Update your Debian 10 installation: ```apt-get update && apt-get upgrade```

dooley1981, 2021-10-03:
okay thanks that did it.

sonnyxp2008, 2021-10-04:
keeps giving the same problem debian 10

sonnyxp2008, 2021-10-04:
aruba cloud

sonnyxp2008, 2021-10-04:
same result GnuTLS: Resource temporarily unavailable, try again.
GnuTLS: Resource temporarily unavailable, try again.
GnuTLS: Resource temporarily unavailable, try again.
ERROR: The certificate of ‘www.letsencrypt.org’ is not trusted.
ERROR: The certificate of ‘www.letsencrypt.org’ has expired.

Ysurac, 2021-10-04:
Seems to fix the issue: ```dpkg-reconfigure ca-certificates``` and remove "DST Root CA X3"

sonnyxp2008, 2021-10-04:
I'm sorry I think I didn't understand, once I delete the certificates as I delete DST Root CA X3 I'm sorry but I can't

sonnyxp2008, 2021-10-04:
This package installs common CA (Certificate Authority) certificates in
/usr/share/ca-certificates. . Please select the certificate authorities
you trust so that their certificates are installed into /etc/ssl/certs.
They will be compiled into a single /etc/ssl/certs/ca-certificates.crt
file.

Certificates to activate:

   [*] mozilla/ACCVRAIZ1.crt
   [*] mozilla/AC_RAIZ_FNMT-RCM.crt
   [*] mozilla/Actalis_Authentication_Root_CA.crt
   [*] mozilla/AddTrust_External_Root.crt
   [*] mozilla/AffirmTrust_Commercial.crt
   [*] mozilla/AffirmTrust_Networking.crt

Ysurac, 2021-10-04:
Try: ```sed -i '/^mozilla\/DST_Root_CA_X3/s/^/!/' /etc/ca-certificates.conf && update-ca-certificates -f```
Worked on a server with the problem I have. If it's not working I will get a server on aruba cloud.

sonnyxp2008, 2021-10-04:
Get:2 http://ftp.uk.debian.org/debian buster InRelease [122 kB]
Get:3 http://ftp.uk.debian.org/debian buster-updates InRelease [51.9 kB]
Reading package lists... Done
E: Repository 'http://security.debian.org/debian-security buster/updates InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
N: Repository 'http://ftp.uk.debian.org/debian buster InRelease' changed its 'Version' value from '10.1' to '10.10'
E: Repository 'http://ftp.uk.debian.org/debian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
E: Repository 'http://ftp.uk.debian.org/debian buster-updates InRelease' changed its 'Suite' value from 'stable-updates' to 'oldstable-updates'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
root@ARU-278058eme:~#

sonnyxp2008, 2021-10-04:
ok after clean installation the installation seems to have started. Thank you

closed 2022-01-07 by github-actions[bot]