Ysurac/openmptcprouter #2110: Unable to update Letsencrypt SSL

hairy-dog, 2021-10-05:
## Expected Behavior

I want to be able to contact letsencrypt.org to update SSL certificates

## Current Behavior
This is the error:

 Current SSL set 6: signed by /C=US/O=Let's Encrypt/CN=R3, expires 2021-11-02 05:26:48 UTC
        The current certificate expires in 28 days.
        Fetching a new certificate from LetsEncrypt.
        !! Failed: Failed to open TCP connection to acme-v02.api.letsencrypt.org:443 (getaddrinfo: Name or service not known)

Yet, it can reach that address:
$ ping acme-v02.api.letsencrypt.org
PING ca80a1adb12a4fbdac5ffcbc944e9a61.pacloudflare.com (172.65.32.248) 56(84) bytes of data.
64 bytes from 172.65.32.248 (172.65.32.248): icmp_seq=1 ttl=58 time=17.8 ms
64 bytes from 172.65.32.248 (172.65.32.248): icmp_seq=2 ttl=58 time=20.0 ms
64 bytes from 172.65.32.248 (172.65.32.248): icmp_seq=3 ttl=58 time=18.3 ms

and when I try

 $ wget -O - https://acme-v02.api.letsencrypt.org/directory

I get 

--2021-10-05 14:56:11--  https://acme-v02.api.letsencrypt.org/directory
Resolving acme-v02.api.letsencrypt.org (acme-v02.api.letsencrypt.org)... 172.65.32.248, 2606:4700:60:0:f53d:5624:85c7:3a2c
Connecting to acme-v02.api.letsencrypt.org (acme-v02.api.letsencrypt.org)|172.65.32.248|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 658 [application/json]
Saving to: ‘STDOUT’

-                                   0%[                                                            ]       0  --.-KB/s               {
  "keyChange": "https://acme-v02.api.letsencrypt.org/acme/key-change",
  "meta": {
    "caaIdentities": [
      "letsencrypt.org"
    ],
    "termsOfService": "https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf",
    "website": "https://letsencrypt.org"
  },
  "newAccount": "https://acme-v02.api.letsencrypt.org/acme/new-acct",
  "newNonce": "https://acme-v02.api.letsencrypt.org/acme/new-nonce",
  "newOrder": "https://acme-v02.api.letsencrypt.org/acme/new-order",
  "revokeCert": "https://acme-v02.api.letsencrypt.org/acme/revoke-cert",
  "u-HehYwGO5c": "https://community.letsencrypt.org/t/adding-random-entries-to-the-directory/33417"
-                                 100%[===========================================================>]     658  --.-KB/s    in 0.002s

2021-10-05 14:56:12 (428 KB/s) - written to stdout [658/658]

## Specifications

  - OpenMPTCProuter version:  0.58.5
  - OpenMPTCProuter VPS version:  5.4.100
  - OpenMPTCProuter VPS provider: Bitfolk
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-10-05:
You can try the last solution I added in #2099

hairy-dog, 2021-10-05:
That didn't work for me. However, I didn't reboot the pi4 afterwards. Should I have done?

closed 2021-10-05 by hairy-dog