Ysurac/openmptcprouter #2112: What is the protocol used to detect and show LTE icon and signal strenght from ethernet modems?

bobafetthotmail, 2021-10-05:
## Expected Behavior

Each modem gets its own LTE and signal strenght indicator

## Current Behavior

I'm using two LTE modem-routers as WAN1 and WAN2. Connected over ethernet to the openMPTCProuter.

If I go to this page  /cgi-bin/luci/admin/system/openmptcprouter/status (the page where I see the Network Overview), I see that one modem (a huawei B310s-22 ) also gets a little LTE signal indicator and a "LTE" to indicate that it is connected over LTE frequency.

The other modem is a Mikrotik Chateau LTE12, and I don't see the icon or signal strenght.

I know this is a cosmetic thing, but can you tell me what protocols is openMPTCProuter using to detect this information? The Mikrotik device has many settings, maybe I can change or enable something so it exposes the right information.

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version:  v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: VM on x86 hypervisor

Ysurac, 2021-10-05:
To detect the information for Huawei, the huawei public API is used. This work on most Huawei router.
Information is also available when QMI or ModemManager are used.
I don't own a Mikrotik modem and I don't know if there is any public API that work on all models.

bobafetthotmail, 2021-10-05:
Ah, bummer. Thanks for answering my curiosity.

All Mikrotik devices able to use a modem have the "same" firmware (RouterOS) so while device CPU architecture may be different, the firmware uses the same web and commandline interfaces, similar to how OpenWrt/DD-Wrt also work. Only their managed switches use a different firmware.

Mikrotik RouterOS has a rich API (it's basically their CLI interface in API form) and newer firmware versions (like on the Chateau) also add REST API support, see the documentation here https://help.mikrotik.com/docs/display/ROS/API
From what I read from there, both APIs require a login/password before you send any other command, so the device owner will have to create a new non-admin user on the device for this to work. It's probably too much fuss just to get an icon on a webpage.

As I said this is mostly cosmetic, I don't really need any of this.

closed 2021-10-05 by bobafetthotmail