Ysurac/openmptcprouter #2113: Step by step instructions to add Wireguard VPN

hairy-dog, 2021-10-06:
## Expected Behavior
I want to add a VPN so that clients on the internet can connect to my LAN through the OMR connection.

I know that this is possible from the response to https://github.com/Ysurac/openmptcprouter/issues/1874 but I do not understand enough to make use of that information.

I would like to know what I need to do to set up a connection either using the VPN inside the OMR router, or on the VPS, or on a server on the LAN.

I rely on instructions that don't assume I know the answer already.

## Current Behavior
OMR is working just fine. I have tried adding PiVPN running wireguard, but I don't know what settings to change to make that work

## Specifications

  - OpenMPTCProuter version: 0.85.5
  - OpenMPTCProuter VPS version: 1026 5.4 100
  - OpenMPTCProuter VPS provider: bitfolk
  - OpenMPTCProuter platform: RPi4

Ysurac, 2021-10-10:
It's outside the scope of OpenMPTCProuter.
A wireguard server is already installed on VPS side, you can follow official Wireguard documentation and use the configuration or create a new one.

hairy-dog, 2021-10-10:
So you are saying that wireguard is already installed and configured on the VPS, but we should follow the instructions on the Wireguard site to install and configure Wireguard on the VPS.
  
That doesn't make any sense, does it?   
  
What is needed is information about what needs to be done on the VPS when OMR is already installed. How the steps differ from the standard install.
  

  

  
  
 -- Iain

highfreq, 2021-10-10:
hairy-dog, OMR is a piece of software for initiated people, it is very chic with its 10 pages wiki and criptic answers by the dev :)

Ysurac, 2021-10-10:
To be clear I'm the only, if you want to make documentation I will add to wiki, but for now I prefer to make the system work.
A VPN on the VPS is not related at all with aggregation. I will not make Wireguard documentation...

highfreq, 2021-10-10:
I was just joking :)

Ysurac, 2021-10-10:
@highfreq I was answering to @hairy-dog ;)
And I have not the time to make a how to for every user special case. I can help but I will not do the work :)

hairy-dog, 2021-10-10:
So are you saying that I could add a whole new wireguard installation on the VPS without upsetting the existing wireguard already installed there?

Ysurac, 2021-10-10:
Wireguard is already installed, so you can add a new configuration, without using wg0 that is currently used and using wg1 for example.
Current wireguard configuration can also be used if port are opened to the internet on Shorewall firewall (/etc/shorewall/rules) and MPTCP over VPN is not used in the wizard of OpenMPTCProuter.

To make something really clean, you can copy /etc/wireguard/wg0.conf to /etc/wireguard/wg1.conf (or what you want), change ListenPort to something not used by OpenMPTCProuter, change Address to 10.255.230.1/24 for example, open the port in /etc/shorewall/rules, generate new keys: ```wg genkey | tee vpn-server-privatekey-wg1.key | wg pubkey > vpn-server-public-wg1.key```, modify path to the key in  wg1.conf and start it (systemctl enable wg-quick@wg1, systemctl start wg-quick@wg1).

hairy-dog, 2021-10-10:
That's really helpful, thanks. I'll get brave and try it soon!

closed 2022-01-17 by github-actions[bot]