Ysurac/openmptcprouter #2124: port forwarding for vpn server does not work but with V2ray it works

bobafetthotmail, 2021-10-15:
## Expected Behavior

configured port forwarding as https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding works

## Current Behavior

The VPN server at IP 192.168.100.2 I'm trying to expose on UDP 64414 is not accessible.
This is firewall configuration on OpenMPTCProuter 
```
config redirect
	option target 'DNAT'
	option name 'ovpnserver'
	list proto 'udp'
	option dest 'lan'
	option dest_ip '192.168.100.1'
	option src_dport '64414'
	option dest_port '64414'
	option src 'vpn'

config nat
	option name 'ovpnserver'
	list proto 'udp'
	option src 'lan'
	option src_port '64414'
	option dest_ip '192.168.100.2'
	option dest_port '64414'
	option target 'SNAT'
	option snat_ip '192.168.100.1'
```

I see ports are closed on VPS and openMPTCP router If I use this command
```
nmap -sU  -p64414 localhost 

PORT      STATE  SERVICE
64414/udp closed unknown
```

But if I switch from shadowproxy to v2ray and I follow the instructions that use V2ray https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding-using-v2ray the port forward works and I can access the VPN server

also the same command reports that the port is now "open|filtered"

```
PORT      STATE         SERVICE
64414/udp open|filtered unknown
```

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026 on Debian 10
  - OpenMPTCProuter VPS provider: Hertzner
  - OpenMPTCProuter platform: x86_64 VM in a Proxmox host

treyguitar, 2021-11-12:
Did you solve it on your own? It's happening to me too, I upgraded VPS and Router and port forwards do not work anymore

Ysurac, 2021-11-12:
@treyguitar open your own bug. But you may need to validate the wizard again to send back info, if not working check advanced checkbox in wizard and check "Force retrieve settings".

EticWeb, 2021-11-12:
It's maybe the same problem I report on lafibre forum. About the IP mention in the wiki for port forwarding. The ip need to be the destination server and not the openmptcprouter as mention in the actual wiki screen shot. I hope it can help.

closed 2022-02-15 by github-actions[bot]