Ysurac/openmptcprouter #2141: Gateway Metric of an interface

MoltesDu12, 2021-10-30:
Hello.
I connected an usbstick for 4G connection to my Raspberry pi. The IP address of this interface is given through a dhcp process I did find.
I would like to use this interface only when all of the others are full. I tried to put a metric to very high value (about 50).
But this value isn't stored definitively : I have to put this value almost every day.
How can I set metric value to be always at the right value ?

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024 5.4.89-mptcp
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI3

Ysurac, 2021-10-30:
Connections are used based on latency only.

MoltesDu12, 2021-10-31:
I've already tried to use latency in order to make this stick priority higher, but this stick is too often used, even if the other connections aren't full. I thought that metric could make using this stick would be only when all of the other uplinks are full.
I just want to know how to make gateway metric fixed.

Ysurac, 2021-10-31:
You can't fix metric and this wouldn't change anything. Latency is the only setting you can change but it was here only for some testing.
Current release doesn't support connections priority, maybe in a future release or by using netlink and a custom mptcpd plugin.

MoltesDu12, 2021-11-02:
Thanks for your answer.
But even if I can't do what I expected using metric gateway, how can I set it definitively ?

closed 2022-02-05 by github-actions[bot]