Ysurac/openmptcprouter #2142: Create a vpn server on the vps on which the bonding is supported

bluenor, 2021-10-30:
Hello I use your fantastic system with pleasure and satisfaction and I congratulate you on how you have implemented it. 

I have a question if it's not too much trouble.

In the settings of my raspberry 4 that I use, I read that there is the possibility to set up an openvpn server. If I understand correctly, through these settings, it might be possible to connect to the vpn that the 
openmptcprouter relies on to aggregate connections and that they come out of it.

My goal is to connect to the vpn where the aggregated connection comes out of from any client that has openvpn. 

Is this possible? Is there a guide? Thank you.

Ysurac, 2021-11-01:
There is a guide here: https://github.com/Ysurac/openmptcprouter/issues/2123

hairy-dog, 2021-11-04:
I'm one of the many other people who want to do this. From what I can see, #2123 ended with the person not being able to get it to work, and asking for more help, without response. Was he trying to do something different?

highfreq, 2021-11-06:
I am just trying to have OMR network to talk to a new wireguard tunnel i created, at this point i can ping the server from my new tunnel client and the client from the server but not able to been rotated to the OMR lan subnet.
Tried ACCEPT in shorewall policy and added a static router on OMR but to no avail at the moment.

closed 2022-02-09 by github-actions[bot]