Ysurac/openmptcprouter #2144: can't get public ip address from shadowsocks

HWCODEING, 2021-11-01:
hi~
i'm from China.
i'm build openmptcprouter for NXP LS1043A(ARM Cortex-A53),Not in the list you provided.

There are some problems：
1. VPN->Glorytun TCP is empty,not have default information.
[image]
2.because Glorytun TCP is empty,so i chose mlvpn,but connect to server is faile.
[image]

## Specifications
  - OpenMPTCProuter version: (v0.58.5)
  - OpenMPTCProuter VPS version: (i don't know get  VPS version)
  - OpenMPTCProuter VPS provider:()
  - OpenMPTCProuter platform:  (NXP LS1043A)

Ysurac, 2021-11-01:
There is no wan configured, so this can't work.

HWCODEING, 2021-11-02:
> There is no wan configured, so this can't work.
[image]
### **### and systemlog.**
Mon Nov  1 10:17:38 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:17:45 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:17:49 2021 user.notice post-tracking-post-tracking: Can't find omr-bypass rules, restart omr-bypass...
Mon Nov  1 10:17:49 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Nov  1 10:17:51 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  1 10:17:52 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  1 10:17:52 2021 daemon.info dnsmasq[18596]: read /etc/hosts - 4 addresses
Mon Nov  1 10:17:52 2021 daemon.info dnsmasq[18596]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  1 10:17:52 2021 daemon.info dnsmasq-dhcp[18596]: read /etc/ethers - 0 addresses
Mon Nov  1 10:17:52 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:17:54 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:17:55 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Mon Nov  1 10:17:55 2021 user.notice Shadowsocks: Rules UP
Mon Nov  1 10:17:56 2021 user.notice Shadowsocks: Rules not applied
Mon Nov  1 10:17:56 2021 user.notice Shadowsocks: Reload omr-bypass rules
Mon Nov  1 10:17:56 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  1 10:17:57 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  1 10:17:57 2021 daemon.info dnsmasq[18596]: read /etc/hosts - 4 addresses
Mon Nov  1 10:17:57 2021 daemon.info dnsmasq[18596]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  1 10:17:57 2021 daemon.info dnsmasq-dhcp[18596]: read /etc/ethers - 0 addresses
Mon Nov  1 10:17:57 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:18:03 2021 user.notice post-tracking-post-tracking: Can't find omr-bypass rules, restart omr-bypass...
Mon Nov  1 10:18:03 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Nov  1 10:18:04 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:18:05 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  1 10:18:07 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  1 10:18:07 2021 daemon.info dnsmasq[18596]: read /etc/hosts - 4 addresses
Mon Nov  1 10:18:07 2021 daemon.info dnsmasq[18596]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  1 10:18:07 2021 daemon.info dnsmasq-dhcp[18596]: read /etc/ethers - 0 addresses
Mon Nov  1 10:18:07 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:18:14 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:18:19 2021 user.notice post-tracking-post-tracking: Can't find omr-bypass rules, restart omr-bypass...
Mon Nov  1 10:18:19 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Nov  1 10:18:20 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  1 10:18:21 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  1 10:18:21 2021 daemon.info dnsmasq[18596]: read /etc/hosts - 4 addresses
Mon Nov  1 10:18:21 2021 daemon.info dnsmasq[18596]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  1 10:18:21 2021 daemon.info dnsmasq-dhcp[18596]: read /etc/ethers - 0 addresses
Mon Nov  1 10:18:21 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:18:23 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:18:32 2021 user.notice post-tracking-post-tracking: Can't find omr-bypass rules, restart omr-bypass...
Mon Nov  1 10:18:33 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Nov  1 10:18:33 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:18:34 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  1 10:18:35 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Mon Nov  1 10:18:35 2021 user.notice Shadowsocks: Rules UP
Mon Nov  1 10:18:35 2021 user.notice Shadowsocks: Rules not applied
Mon Nov  1 10:18:35 2021 user.notice Shadowsocks: Reload omr-bypass rules
Mon Nov  1 10:18:35 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  1 10:18:37 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  1 10:18:37 2021 daemon.info dnsmasq[18596]: read /etc/hosts - 4 addresses
Mon Nov  1 10:18:37 2021 daemon.info dnsmasq[18596]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  1 10:18:37 2021 daemon.info dnsmasq-dhcp[18596]: read /etc/ethers - 0 addresses
Mon Nov  1 10:18:37 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:18:38 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  1 10:18:38 2021 daemon.info dnsmasq[18596]: read /etc/hosts - 4 addresses
Mon Nov  1 10:18:38 2021 daemon.info dnsmasq[18596]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  1 10:18:38 2021 daemon.info dnsmasq-dhcp[18596]: read /etc/ethers - 0 addresses
Mon Nov  1 10:18:38 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:18:43 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:18:50 2021 user.notice post-tracking-post-tracking: Can't find omr-bypass rules, restart omr-bypass...
[... cut ...]
Mon Nov  1 10:19:01 2021 daemon.info dnsmasq[18596]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  1 10:19:01 2021 daemon.info dnsmasq-dhcp[18596]: read /etc/ethers - 0 addresses
Mon Nov  1 10:19:01 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  1 10:19:03 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.253.1 dev mlvpn0
Mon Nov  1 10:19:04 2021 user.notice post-tracking-post-tracking: Can't find omr-bypass rules, restart omr-bypass...
Mon Nov  1 10:19:04 2021 user.notice post-tracking-post-tracking: Set firewall on server

HWCODEING, 2021-11-02:
hi
I have new developments here.but i have one error. server ip addr is no dispaly.
i chose V2Ray  mlvp adn china county
[image]
no dispaly information
[image]
@Ysurac

closed 2022-02-05 by github-actions[bot]