Ysurac/openmptcprouter #2149: Install Mikrotik Re11-LTE Modem

GTiDon, 2021-11-02:
Hello

Please help to install this modem
Mikrotik Re11-LTE Modem

## Current Behavior
Currently I can detect the modem but OpenWRT says it is disabled.
I tried using it with Modem Manager, UMTS/GPRS/EV-DO and NCM

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - 
  - ease do not attach text files) -->

GTiDon, 2021-11-02:
[image]
[image]

GTiDon, 2021-11-02:
[image]

Nickmcl024, 2022-01-17:
Any luck get getting these modems to work with it

closed 2022-04-22 by github-actions[bot]