Ysurac/openmptcprouter #2150: How to reset OpenMPTCPRouter wan addresses

excel4, 2021-11-02:
## Expected Behavior
I am using OpenMPTCPRouter (OMR) with two LTE links. The links change the wan address  periodically.  I was expecting OMR to recognize this change and adjust the routes.

## Current Behavior:
When the wan IP address for wan1 changes (master), OMR goes down for some time and then restores service. 
However, when the wan IP address for wan2 changes, OMR continues to attempt to use the previous (invalid) wan2 IP address and reports no server ping response on wan2 and no latency. wan2 is thus not part of the bond.

I rebooted the router and VPS, however, the stale wan address remains.
I also swapped the Multipath TCP master and enable roles, however the stale wan address remains.
Eventually wan2 seems to come back on line, but it goes off line again when the address changes. I have verified that the wan2 link is actually working.

## Question:
Is there a setting that enables/forces OMR to reset/refresh the wan address when it fails to respond?

## Specifications
OMR: v0.58.5 on WRT3200ACM
VPS version 0.1026 on Debian 20.04

Configuration note: OMR has a fixed private address for wan1 and wan2 from an intermediate router. OMR reports The LTE device wan addrewss as 172.58.x.x. However, traceroute from OMR reports a 10.198.x.x address as the next hop after the LTE modem lan address and shows no 172.58.x.x address. So I don't know how to determine the 172.58 address or how OMR determines that.

Ysurac, 2021-11-05:
Wan public IP is only used for display in status page. It should be updated after some time.
The "no server ping response on wan2" really mean there is no server ping answer. ping is bind to wan local ip to ping server ip, the result is also used to display latency.

excel4, 2021-11-06:
Thanks for the update. Let me clarify. the OMR router stops routing traffic on wan2 after some time. However, wan2 is alive and able to reach the internet. The path is:
OMR/192.168.102.1 -> R1/192.168.20.1 -> R2/192.168.120.1 -> R3/192.168.220.1
R3 is the LTE device. R1 and R2 are intermediary routers.
The OMR routing table includes routes to R2 and R3.

* when working, R1, R2, R3 and the internet respond ping from the OMR router.
* when failing, R1 and R2 respond to OMR. R3 is not reachable. The internet is also not reachable on wan2.
-- however, from R1, R2, R3 and the internet respond. So the issue is local to OMR.

Are there some additional diagnostics I can run to identify the problem?

santosh07bec, 2022-01-31:
I am also facing same issue. To be specific, here is how my setup looks like:
MacLaptop(my workstation) -> WifiRouter -> OMR ->| ->   eth1  -> LTE Router1 ->| VPS -> Internet
                                                                                           | -> wlan0 -> LTE Router2 ->|
When on the luci status page wlan0 status shows warning, I see ICMP ping going out via interface wlan0 and reaching VPS interface with WAN_IP1 and response sent back from VPS, but on the luci status page the WAN IP displayed is different (say WAN_IP2) and I do not get icmp response back on wlan0 interface.

However, when on the luci status page wlan0 status shows green, I see icmp ping going out to VPS and Coming back with same WAN_IP displayed on the luci status page.

While the luci shows warning on the status page, if I connect to the LTE Router2 directly, I can ping VPS and browse internet without any issues.

So this indicates when the OMR detected WAN IP is different than the actual WAN ip for the LTE connection, ICMP ping fails due to some reason and makes that LTE connection unstable.

Any advise on how to fix this issue?

closed 2022-05-06 by github-actions[bot]