Ysurac/openmptcprouter #2154: Linksys WRT3200ACM interfaces

applamosca, 2021-11-03:
on the Linksys WRT3200ACM i can't set interfaces interfaces in bridge someone can help me
[image]

Ysurac, 2021-11-05:
Please use template.
If you can't set interface in bridge mode it's because you can't, it's not supported on this platform.

applamosca, 2021-11-05:
I am following this video tutorial and it tells me to install this firmware, then to set the lan port in bridge.
Where am I wrong? can you help me I'm your supporter
[image]
[image]

closed 2022-02-09 by github-actions[bot]