Ysurac/openmptcprouter #2156: Excessive DNS and HTTPS request number every second directed to data.iana

jjak0b, 2021-11-03:
## Expected Behavior
I don't expect the router perform to many DNS and HTTPS requests every second just to fetch files from iana

## Current Behavior
The router perform (i belive approximately) every second some DNS requests to a DNS server set by me which is a opendns address.
But i figured out that these DNS requests are used to fetch the ip address and get a file from data.iana because 72.21.81.189 match with that domain domain where you can find this file [root-anchors.xml](https://data.iana.org/root-anchors/root-anchors.xml) or some other files related to that URI.

## Steps to Reproduce the Problem

  1. Install nlbwmon
  2. Don't use internet with any device
  3. You can see from nlbwmon monitor and realtime graphs a lot of DNS connections requested by the master interface

## Context (Environment)
Is this a correct behavior ?
I discovered this because in 1 month i tracked the router traffic with nlbwmon and the master interface used a lot of GBytes of DNS traffic. It's too much considering that the master interface is a 4G LTE network with limited GBytes of internet traffics. ( My master interface have ip 192.168.1.252 .
I tried also on changing master interface and the excessive traffic is still there but on the changed master interface ip.

[image]

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter VPS provider: N/A
  - OpenMPTCProuter platform: RPI4
  - Country: Italy

Ysurac, 2021-11-05:
Check if you filesystem is not read only.
How did you set your custom DNS ?

jjak0b, 2021-11-05:
The filesystem is not read only.
I set the OpenDNS in Network ->DHCP and DNS page: 
[image]

In syslog there are the also the same following messages every 30 seconds since boot, i don't know if this could be related to this issue

```
...
Fri Nov  5 12:31:14 2021 user.notice post-tracking-post-tracking: Set firewall on server
Fri Nov  5 12:31:22 2021 user.notice omr-bypass: Reload dnsmasq...
Fri Nov  5 12:31:22 2021 daemon.info dnsmasq[31469]: read /etc/hosts - 4 addresses
Fri Nov  5 12:31:22 2021 daemon.info dnsmasq[31469]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Fri Nov  5 12:31:22 2021 daemon.info dnsmasq-dhcp[31469]: read /etc/ethers - 0 addresses
Fri Nov  5 12:31:22 2021 user.notice omr-bypass: OMR-ByPass is running
Fri Nov  5 12:31:22 2021 user.notice omr-bypass: Starting OMR-ByPass...
Fri Nov  5 12:31:41 2021 user.notice post-tracking-post-tracking: Set firewall on server
Fri Nov  5 12:31:50 2021 user.notice omr-bypass: Reload dnsmasq...
Fri Nov  5 12:31:50 2021 daemon.info dnsmasq[31469]: read /etc/hosts - 4 addresses
Fri Nov  5 12:31:50 2021 daemon.info dnsmasq[31469]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Fri Nov  5 12:31:50 2021 daemon.info dnsmasq-dhcp[31469]: read /etc/ethers - 0 addresses
Fri Nov  5 12:31:50 2021 user.notice omr-bypass: OMR-ByPass is running
Fri Nov  5 12:31:50 2021 user.notice omr-bypass: Starting OMR-ByPass...
```

jjak0b, 2021-11-05:
fs is not read only because i wrote an auto-repair fs code in System->Startup -> Local Startup, because on power failure the fs was always corrupted (even if on SD or USB) and went to into read only.
```
# if read-only: check, repair and reboot 
dane=$(mount | grep '/dev/root' | awk '{print $6}' )
if [ "$dane" != "(rw,noatime)" ]
then
fsck.ext4 -y /dev/sda2
reboot
fi
```

I wrote this because the fs don't auto repair fs errors even if specific flags are enabled

[image]

Ysurac, 2021-11-05:
I think the DNS problem is related to previous log.
Can you give me, via SSH on the router, the result of ```uci show openmptcprouter | grep set_firewall``` et de ```iptables-save | grep omr_dst_bypass``` ?

jjak0b, 2021-11-05:
-  uci show openmptcprouter | grep set_firewall

```
openmptcprouter.vps.set_firewall='1'
```

- iptables-save | grep omr_dst_bypass

```
-A omr-bypass -m set --match-set omr_dst_bypass_usb0 dst -j MARK --set-xmark 0x5397/0xffffffff
-A omr-bypass -m set --match-set omr_dst_bypass_tun0 dst -j MARK --set-xmark 0x5391200/0xffffffff
-A omr-bypass -m set --match-set omr_dst_bypass_wan2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A omr-bypass -m set --match-set omr_dst_bypass_wan1 dst -j MARK --set-xmark 0x5393/0xffffffff
-A omr-bypass -m set --match-set omr_dst_bypass_eth0 dst -j MARK --set-xmark 0x5396/0xffffffff
-A omr-bypass -m set --match-set omr_dst_bypass_lo dst -j MARK --set-xmark 0x5395/0xffffffff
-A omr-bypass -m set --match-set omr_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
-A omr-bypass-local -m set --match-set omr_dst_bypass_usb0 dst -j MARK --set-xmark 0x5397/0xffffffff
-A omr-bypass-local -m set --match-set omr_dst_bypass_tun0 dst -j MARK --set-xmark 0x5391200/0xffffffff
-A omr-bypass-local -m set --match-set omr_dst_bypass_wan2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A omr-bypass-local -m set --match-set omr_dst_bypass_wan1 dst -j MARK --set-xmark 0x5393/0xffffffff
-A omr-bypass-local -m set --match-set omr_dst_bypass_eth0 dst -j MARK --set-xmark 0x5396/0xffffffff
-A omr-bypass-local -m set --match-set omr_dst_bypass_lo dst -j MARK --set-xmark 0x5395/0xffffffff
-A omr-bypass-local -m set --match-set omr_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
```

Ysurac, 2021-11-05:
So the problem is "openmptcprouter.vps.set_firewall='1'": this mean firewall can't be correctly set on VPS side.
I would need the result of ```uci show firewall``` from the router and the contain of file ```/etc/shorewall/rules``` from the VPS.

jjak0b, 2021-11-05:
I have no vps set on the router yet. I'm using the router for the OMR-bypass functionality to use my primary 4G network for normal use as master and another ADSL network (as enabled) to use it as "interface for downloads " for services stuff like steam, amazon prime, etc ...

- uci show firewall
```
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].input='REJECT'
firewall.@defaults[0].output='REJECT'
firewall.zone_lan=zone
firewall.zone_lan.name='lan'
firewall.zone_lan.network='lan'
firewall.zone_lan.input='ACCEPT'
firewall.zone_lan.output='ACCEPT'
firewall.zone_lan.forward='ACCEPT'
firewall.zone_lan.auto_helper='0'
firewall.zone_lan.mtu_fix='1'
firewall.zone_wan=zone
firewall.zone_wan.name='wan'
firewall.zone_wan.input='REJECT'
firewall.zone_wan.output='ACCEPT'
firewall.zone_wan.forward='REJECT'
firewall.zone_wan.masq='1'
firewall.zone_wan.mtu_fix='1'
firewall.zone_wan.network='wan1' 'wan2' 'USB_PHONE'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].src_ip='fc00::/6'
firewall.@rule[3].dest_ip='fc00::/6'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Forward'
firewall.@rule[5].src='wan'
firewall.@rule[5].dest='*'
firewall.@rule[5].proto='icmp'
[... 120 lines cut ...]
firewall.allow_dhcp_request_vpn.dest_port='67'
firewall.allow_dhcp_request_vpn.target='ACCEPT'
firewall.allow_dhcp_request_vpn.family='ipv4'
firewall.v2ray=include
firewall.v2ray.path='/etc/firewall.v2ray-rules'
firewall.v2ray.reload='0'
firewall.omr_bypass=include
firewall.omr_bypass.path='/etc/firewall.omr-bypass'
firewall.omr_bypass.reload='0'
firewall.miniupnpd=include
firewall.miniupnpd.type='script'
firewall.miniupnpd.path='/usr/share/miniupnpd/firewall.include'
firewall.miniupnpd.family='any'
firewall.miniupnpd.reload='1'
firewall.ss_rules=include
firewall.ss_rules.path='/etc/firewall.ss-rules'
firewall.ss_rules.reload='1'
firewall.@redirect[0]=redirect
firewall.@redirect[0].target='DNAT'
firewall.@redirect[0].name='DMZ'
firewall.@redirect[0].proto='all'
firewall.@redirect[0].src='wan'
firewall.@redirect[0].dest='lan'
firewall.@redirect[0].dest_ip='192.168.1.60'
```

Ysurac, 2021-11-05:
it was not a tested usage, and so it's a bug when no VPS is set.
You can on the router via SSH do this ```uci set openmptcprouter.vps.set_firewall=0 && uci commit``` to fix the issue for now.

jjak0b, 2021-11-05:
Thanks now seems fixed, but i have to manually set it again to 0 after each reboot.

jjak0b, 2022-02-05:
has it been fixed yet ?

closed 2022-05-11 by github-actions[bot]