Ysurac/openmptcprouter #2157: Install on PC

mvmaksnet, 2021-11-04:
I use image x86 (32-bits):
ext4 image: openmptcprouter-v0.58.5-r0+16336-b36068d35d-x86-generic-generic-ext4-combined.img.gz (2021/07/10 - 69.30Mo - sha256sum)

I use balenaetcher for making hdd partition from image.
But when I boot pc I don't getting IP address to PC and also can't get to 192.168.100.1 web gui.
Am I wrong in process of installation?

killereiz, 2021-11-04:
Well do you have a monitor installed? Have you checked in the bios that the correct drive is booting? Upon boot you should see if the interfaces brings up the network interface.

mvmaksnet, 2021-11-05:
I tried access to router through wrong eth port. It is working. 
I have question for you. How can I access web gui  of router from anywhere?

Adorfer, 2021-11-05:
1) if you issue is resolved: close it please
2) if you have a different issue/question that is not related to "install on PC", a separate issue would be appropriate.  My first guess is: It's not a bug/issue, you simply have to configure that. it's working here. But solution/method is out of scope of this issue.

mvmaksnet, 2021-11-08:
It is working. It was my mistake.

closed 2021-11-08 by mvmaksnet