Ysurac/openmptcprouter #2160: Download loss problem

JulesCatherine, 2021-11-07:
## Expected Behavior

I use a Livebox ADSL 15 / 0.8Mbps plus Free 4G 280 / 28Mbps and SFR Red 4G 40 / 40Mbps which makes a total of 335 in download we could say 300 with some loss. (by google translate)

<J'utilise une Livebox ADSL 15/0.8Mbps + Forfait free 4G 280/28Mbps + Forfait SFR Red 4G 40/40Mbps ce qui fait un total de 335 en download on pourrait dire 300 avec une certaine perte.>

## Current Behavior

With its 3 connections I reach speeds on Nperf which turns more around 170 / 60Mbps 

<Avec ses 3 connexions j'atteint des débits sur Nperf qui tourne plus autour de 170/60Mbps>

I have all checked everything is green on the router I tested with 2 different VPS OVH comfort version with 1Gb of bandwidth and Sapinet package 3 € 10Gb of bandwidth. I even created different VMs. (google translate)

<j'ai tous vérifié tout est vert sur le routeur j'ai testé avec 2 VPS différent OVH version confort à 1Gb de bande passante et Sapinet forfait 3€ 10Gb de bande passante. J'ai même crée différentes VM>

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026.5.4.100/ j'ai pas regardé mais installé via le script le 07/11/21
  - OpenMPTCProuter VPS provider: OVH/Sapinet
  - OpenMPTCProuter platform: Virtualbox : 64bit, 4CPU, 8GoRam, Serveur intel Pro/1000 MT

First of all thank you Ysurac you are doing a crazy job !!
So I would like to know if it was possible to optimize the router because I do not see where the problem can come from because I find that despite the encoding and the potential loss I find that it is a lot. The problem of the VPS can in my opinion be ruled out because tested OVH and Sapinet same results.
The case of the VM tested different VMs after virtualbox forced me to a potential clamping?
I tested my local network the figures are good and all my cables are category 7 minimum.

Thank you for your future help. (by google translate)

<Tout d'abord merci Ysurac tu fais un travail de ouf !!
Donc j'aimerai savoir si il était possible d'optimiser le routeur car la je ne vois d’où le problème peut venir car je trouve que malgré de l'encodage et une potentiel perte je trouve que ça fait beaucoup. Le problème du VPS peut à mon avis être écarté car testé OVH et Sapinet même résultats.
Le cas de la VM testé différentes VM après est ce que virtualbox me contraint à un potentiel bridage ?
j'ai testé mon réseau en local les chiffres sont bon et tous mes câbles sont de catégories 7 minimum.

Merci de votre futur aide>

Ysurac, 2021-11-07:
You need to give me the result, via SSH on the router ```omr-test-speed``` for aggregated speed and ```omr-test-speed <interface>``` for each WAN real interface (use ```ip a``` to get them). Test must be run for at least 2 minutes, Ctrl+C to stop.

JulesCatherine, 2021-11-07:
Thank you for your quick response here are my results:
https://drive.google.com/drive/folders/1wkRCnzIGl1QhPe3I8dmZthFRFJ2y8iIe?usp=sharing

Ysurac, 2021-11-08:
There is a problem in current omr-test-speed, OVH banned us to use it ;)
You need to remove the url like this I think: ```sed -i 's#http://proof.ovh.net/files/10Gb.dat ##' /bin/omr-test-speed``` and run it again

JulesCatherine, 2021-11-08:
Even while placing the command, I did not succeed in doing the omr-test-speed with OVH, so I did it with the VPS Sapinet.

I put the screenshot directly in the same folder : https://drive.google.com/drive/folders/1wkRCnzIGl1QhPe3I8dmZthFRFJ2y8iIe?usp=sharing

JulesCatherine, 2021-11-08:
So, by continuing different tests, I wanted to leave my VM and go live with a bootable key. And it seems that this improves my results I get 285/70 Mbps which is much better.
have if that does not change over time.

<Alors, en continuant différent test j'ai voulu quitté ma VM et passer en direct avec une clé bootable. Et il semblerait que cela améliore mes résultats j'obtient 285/70 Mbps ce qui est beaucoup mieux.
avoir si cela ne change pas avec le temps.>

hairy-dog, 2021-11-19:
I was seeing a similar problem (but with slower connections) and could not find a cause. 

To help trace the problem, I enabled just one of the wan connections at a time and ran a speedtest.net speed test on each. 

Then I re-enabled all the wan connections and full speed was restored.

It may help fix your problem, and is very easy to do.

JulesCatherine, 2021-11-22:
Thank you for your answers.
My problem seems to be solved by switching from the virtual machine to a physical machine.

Despite a small loss always but which is understandable on large flow like that.

thank you all

closed 2021-11-22 by JulesCatherine