Ysurac/openmptcprouter #2164: Connection refused every 2 boots

highfreq, 2021-11-08:
OMR OpenMPTCProuter VPS 0.1026
Ubuntu 20.04 GNU/Linux 5.4.100-mptcp x86_64
Raspberry 4
Lately i started having a problem, every 2-3 boots the system doesn't come up and if i try to connect to it i get connection refused. If i restart the raspberry it starts and works normally until next reboot.
Didn't try to reboot server yet.
Here the log:

Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'wifi'
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[9] (Allow-All-LAN-to-VPN) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[12] (Allow-Lan-to-Wan) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[1] (MODEM_1) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[2] (MODEM_2) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[3] (SSH OMR) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[4] (TELEMETRY) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[5] (IONODES WEB) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[6] (IONODES RTSP) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[7] (SSH GIMBAL) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[8] (SSH MODEM_1) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[9] (SSH MODEM_2) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[10] (Basecam) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[11] (Supervisor) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[13] (SSH_TELEMETRY) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @redirect[14] (COCKPIT) does not specify a protocol, assuming TCP+UDP
Mon Nov  8 17:59:09 2021 daemon.notice wpa_supplicant[3392]: Successfully initialized wpa_supplicant
Mon Nov  8 17:59:09 2021 user.notice : Added device handler type: 8021ad
Mon Nov  8 17:59:09 2021 user.notice : Added device handler type: 8021q
Mon Nov  8 17:59:09 2021 user.notice : Added device handler type: macvlan
Mon Nov  8 17:59:09 2021 user.notice : Added device handler type: veth
Mon Nov  8 17:59:09 2021 user.notice : Added device handler type: bridge
Mon Nov  8 17:59:09 2021 user.notice : Added device handler type: Network device
Mon Nov  8 17:59:09 2021 user.notice : Added device handler type: tunnel
Mon Nov  8 17:59:10 2021 cron.err crond[3799]: crond (busybox 1.33.0) started, log level 5
Mon Nov  8 17:59:10 2021 kern.warn kernel: [   14.036749] bcmgenet: Skipping UMAC reset
Mon Nov  8 17:59:10 2021 kern.info kernel: [   14.042181] bcmgenet fd580000.ethernet: configuring instance for external RGMII
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'lan' is enabled
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'lan' is setting up now
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'lan' is now up
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan1' is enabled
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan1' is setting up now
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan1' is now up
Mon Nov  8 17:59:10 2021 daemon.notice netifd: macvlan 'wan1' link is up
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan1' has link connectivity
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan2' is enabled
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan2' is setting up now
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan2' is now up
Mon Nov  8 17:59:10 2021 daemon.notice netifd: macvlan 'wan2' link is up
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'wan2' has link connectivity
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'loopback' is enabled
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'loopback' is setting up now
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'loopback' is now up
Mon Nov  8 17:59:10 2021 kern.info kernel: [   14.049852] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Mon Nov  8 17:59:10 2021 daemon.debug dnsmasq[2403]: listening on wan2(#17): 192.168.11.2 port 53
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Nov  8 17:59:10 2021 daemon.debug dnsmasq[2403]: listening on wan1(#16): 192.168.10.2 port 53
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Nov  8 17:59:10 2021 daemon.debug dnsmasq[2403]: listening on eth0(#2): 192.168.144.200 port 53
Mon Nov  8 17:59:10 2021 daemon.debug dnsmasq[2403]: listening on lo(#1): 127.0.0.1 port 53
Mon Nov  8 17:59:10 2021 daemon.debug dnsmasq[2403]: listening on lo(#1): ::1 port 53
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Network device 'lo' link is up
Mon Nov  8 17:59:10 2021 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon Nov  8 17:59:10 2021 daemon.notice netifd: radio0 (4123): command failed: No error information (-524)
Mon Nov  8 17:59:10 2021 user.notice ModemManager: hotplug: add network interface wan1: event processed
Mon Nov  8 17:59:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=wan1, subsystem=net
Mon Nov  8 17:59:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov  8 17:59:10 2021 user.notice ModemManager: hotplug: add network interface wan2: event processed
[... 517 lines cut ...]
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using only locally-known addresses for domain localhost
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using only locally-known addresses for domain local
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using only locally-known addresses for domain invalid
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using only locally-known addresses for domain bind
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using only locally-known addresses for domain use-application-dns.net
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using only locally-known addresses for domain lan
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using nameserver 127.0.0.1#5353
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: using 1 more local addresses
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: read /etc/hosts - 4 addresses
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq[23653]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon Nov  8 18:02:24 2021 daemon.info dnsmasq-dhcp[23653]: read /etc/ethers - 0 addresses
Mon Nov  8 18:02:25 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Nov  8 18:02:27 2021 user.notice omr-bypass: Starting OMR-ByPass...
[... cut ...]
Mon Nov  8 18:02:43 2021 daemon.info unbound: [20813:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Mon Nov  8 18:02:44 2021 daemon.notice unbound: [27901:0] notice: init module 0: validator
Mon Nov  8 18:02:44 2021 daemon.notice unbound: [27901:0] notice: init module 1: iterator
Mon Nov  8 18:02:44 2021 daemon.info unbound: [27901:0] info: start of service (unbound 1.13.1).
Mon Nov  8 18:02:44 2021 daemon.info unbound: [27901:0] info: generate keytag query _ta-4f66. NULL IN

highfreq, 2021-11-10:
Anyone has any idea on what could be the issue here?

Adorfer, 2021-11-12:
the issue seems to be that you refused to use the template. So it is hard to guess what the actual symptom "not coming up" does actually mean.

highfreq, 2021-11-12:
Sorry for not using the template correctly.
What i meant is tht every other boot when i try to connect to the router trough the vps, the connection gets refused.
It looks like the router didn't pass the port forwarding rules to the vps for some reason.

Adorfer, 2021-11-12:
> i try to connect to the router trough the vps, the connection gets refused.

so you try to ssh from the VPS to the OMR and there the connection is refused. thanks for clarification!
I first thoght it was the portforwardings "from internet via vps to ORM to lan" are not initialized and internet-access to services behind OMR get "connection refused", because that would be an issue with the application of the port forwardings and firewall rules.

highfreq, 2021-11-12:
I try ssh from my pc going trough VPS, not from the VPS. So it may be port forwarding.

closed 2022-02-15 by github-actions[bot]