Ysurac/openmptcprouter #2173: Remove firewalls rules trough bash

bagou4502, 2021-11-10:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: i can t get it because openmptcprouter was down
  - OpenMPTCProuter VPS version: i can t get it because openmptcprouter was down
  - OpenMPTCProuter VPS provider: hostmyserver
  - OpenMPTCProuter platform: x86

Hey i add recently a firewall rule on openmptcprouter panel for open a port but after that the panel was unreachable
How  i can remove this rule trough ssh or bash?

Ysurac, 2021-11-11:
All configuration is available via SSH using ```uci```, it's also in /etc/config

bagou4502, 2021-11-11:
yes but in where file i can remove this rule??

bagou4502, 2021-11-11:
what file*

bagou4502, 2021-11-11:
i can t see the openned port in ssh is very strange

Ysurac, 2021-11-11:
```/etc/config/firewall```

bagou4502, 2021-11-12:
> ```/etc/config/firewall```

I see notching about my port modification in this file...

closed 2022-02-15 by github-actions[bot]