Ysurac/openmptcprouter #2177: Max aggregated throughput

sadahsan, 2021-11-12:
## Expected Behavior
I am testing mptcp router and vps (Debian 10) with latest version from the openmptcprouter website on 10G local network and am expecting multi gigabit throughput.

## Current Behavior
When I am running iperf3 between router and VPS IP, I am getting around 8Gbps but when I add another 10G interface on VPS e.g eth2 192.168.110.1 and if I run the test on this interface all traffic is routing via glorytun TCP tunnel and am only getting 1Gbps.

is this expected behaviour or shall I expect to achieve multi gigabit throughput?

sadahsan, 2021-11-13:
Default route on omr is showing via glorytun with metric 0, hence traffic destined to the other side of vps is using glorytun, is this how it should be? Seems like glorytun is struggling hence I am not getting more than 1Gbps but on vps IP i can get 8Gbps.

i would really appreciate your advise, thanks

Ysurac, 2021-11-13:
By default TCP use the proxy shadowsocks, and everything else use the VPN (it's glorytun TCP by default).

sadahsan, 2021-11-13:
Thank you, this is my understanding too. But when I am running tcp iperf test between two devices (one connected to OMR which is getting the IP via DHCP and other connected to VPS) then I am observing traffic is going through VPN (glorytun TCP interface).
On OMR default route is via glorytun TCP (tun0) so device connected on the other side of VPS is only reachable via tun0 interface regardless of TCP or anything else.

Ysurac, 2021-11-13:
TCP traffic is redirected using an iptable rule to shadowsocks.

sadahsan, 2021-11-13:
Thank you, will check that. Is that rule needs to be on both sides i.e. Omr and vps or just on omr?

sadahsan, 2021-11-14:
Hi again, I have below rule on OMR
-A ssr_def_forward -p tcp -j REDIRECT --to-ports 1100-1103
there are also other ss rules on omr but there seems to be something wrong, TCP traffic generated between iperf client (192.168.200.192) and iperf server (192.168.197.10) is still going via glorytun.

[image]

Please see the high load on glorytun 
[image]

Routing table on omr
[image]
[image]

I can't see any errors in system log on OMR

I would really appreciate your help, Thanks.

Ysurac, 2021-11-14:
As you should see in IP route, there is a direct route to VPS IP. It's why you can't test speed between router and VPS.
Only valid speed tests are between internet and router. omr-test-speed command is made for that.

sadahsan, 2021-11-14:
Hi Yannick, route to vps (192.168.197.12) is via two wan interfaces (192.168.201.1 and 192.168.202.1).
When I run the iperf tcp test between two devices, one connected to omr and other connected to vps, traffic is going through glorytun and I expect that to go through shadowsocks, this seems to be the problem.
I will do a drawing for the setup I have.

sadahsan, 2021-11-14:
All working now Yannick, thanks for your help. With my processor I can get around 3.5Gbps on tcp (shadowsocks).

closed 2021-11-14 by sadahsan