Ysurac/openmptcprouter #2179: MacVLAN not working with specific router on RPi 4 internal NIC ethernet?

kamileq, 2021-11-13:
Regarding MacVLAN - is there any hardware of router need to be compatible with Raspberry Pi 4 B ethernet NIC? 

I tried many hours figure out to connect the MoFi Router OLAX MF6875 (I found it by accident, cheap mobile LTE router wifi with ethernet port - that really works!) to RPI ethernet NIC, but without success. Then I plug it, with the same configuration (no matter than is physical eth1 or virtual MacVLAN wan) IP address (of course I disable DHCP and put static IP) to separate USB 3.0 Gigabit Ethernet adapter (KY-888 with RTL8153 chip) and it works - for now MPTCP works fine. 

I also test another 2 routers (D-Link, Asus) on internal RPi 4 ethernet NIC and MacVLAN works fine, but not all time. I do not know why but sometimes the USB 3.0 adapter reset, status in router said about that and connection is reset - looks like power or chip issue? For sure there is no problem with USB-C power adapter - it works fine all time. 

**So question: MacVLAN -  is there any hardware of router need to be compatible with Raspberry Pi 4 B ethernet NIC? (technology? standards?)**

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2711-rpi-4
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1026
  - OpenMPTCProuter VPS provider: aruba
  - OpenMPTCProuter platform: RPI4 64-bit

Ysurac, 2021-11-16:
Not problem with MacVLAN. Only problem can be if connected switch have a filter on mac by port enabled.
If adapter reset it's really in most case a power issue, RPI4 doesn't deliver much power on USB so a powered USB hub is often needed.

kamileq, 2021-11-16:
So why when I tried on external NIC with the same configuration it works
fine?
Switch or router have not enabled “filter on mac by port” - how to check
that? See All settings and not checked. I also check on 3 switches.
I think that is a power issue - but got USB 3.0 the RPi4 gave more power I
read. Need to check some usb 3.0 powered hub or another external NIC

KD

closed 2022-02-20 by github-actions[bot]