Ysurac/openmptcprouter #2183: RaspberryPi 400 - No Wireless interface - Driver issue ?

rmribeiro1, 2021-11-18:
I, 

I just flashed:

Raspberry PI 4B (64-bits)
ext4 image: openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz (2021/07/10 - 73.90Mo - sha256sum)

On a Raspsberry 400 (  I know it is a weird option but I could not find a RP 4 in stock anywhere ). 

It boots ok, seems to be working but there is no wireless option on Luci and I cannot see the interface on the command line. 

Am I missing something? How do I bring up the wireless interface? Is raspberry Pi 400 different to raspberry pi 4  in this field?

rmribeiro1, 2021-11-20:
Quick Update.

While not knowing what to do I download the /lib/firmware/brcm/  from the latest raspbian and then uploaded to the raspberry pi 400 

scp * root@192.168.100.1:/lib/firmware/brcm/ 

Reloaded and now the wireless interface is available in interface and looking at ifconfig it looks like it is working : 

root@OpenMPTCProuter:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr E4:5F:01:36:13:2F  
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3919 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3265 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:442371 (432.0 KiB)  TX bytes:963617 (941.0 KiB)

eth0      Link encap:Ethernet  HWaddr E4:5F:01:36:13:2F  
          UP BROADCAST RUNNING PROMISC ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:3926 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3265 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:497939 (486.2 KiB)  TX bytes:963617 (941.0 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3693 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3693 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:275128 (268.6 KiB)  TX bytes:275128 (268.6 KiB)

wlan0     Link encap:Ethernet  HWaddr E4:5F:01:36:13:30  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:8 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:792 (792.0 B)  TX bytes:2202 (2.1 KiB)

but I still have to test it.  

First try I could not put in AP mode.

Adorfer, 2021-11-22:
Just my 2cents: Raspi4 as client is OKish, but as AP it is REALLY bad. throughput, range and is frustrating, compared even to the cheapest accesses-points in the "sub15 bucks" range. 
It may be usable for some debugging, but never try to push payload with it.

rmribeiro1, 2021-11-23:
I understand. But my objective for using the built in wifi was to build a mobile router that I could take everyware and be battery operated.   So range there would be line of site, and then I could connect to mobile phone ( Client or AP ) depending on the situation.    But thanks for the feedback , makes perfect sense and for a home deployment you let the AP shine.   

On a quick update as was able to turn on: 
  - Client mode only and connect to a 2.4 wifi with no issues 
  -  AP mode only  and connect to it 
  -  Was not able to do 2.4ghz Client and 5ghz AP mode  but did not look into it , might be that you cannot do that on one radio.

closed 2022-02-27 by github-actions[bot]