Ysurac/openmptcprouter #2185: Can't work when VPS is under NAT!

bob7gr, 2021-11-21:
Hello, I tried openmptcprouter on a VPS-VM that I created via Virtualbox on a dedicated server and I was forced to give it an internal address of 10.0.2.15 because Bridged Adapter doesn't work (maybe can't take more public ip). Even if I port forwarded all 65XXX ports matching to all services hosted in the VPS-VM,  on my OpenMPTCPRouter, I still see "Multipath seems to be blocked on the connection" on all my wan interfaces. Also, strange is the fact that instead of observing a real wan address of a connection, I see the 10.0.2.2 virtual wan of the VPS instead. I also tried a VPS from other company with direct public ip and worked there without the above message but I would like to avoid it and stick to my dedicated server and its vms instead of renting another one machine for only this job, thank you!

Ysurac, 2021-11-21:
Please use the template.

Check that VPS-VM have a working internet connection with DNS also working. Check also shorewall configuration and ethernet ip defined in it, it try to use the first interface with a public IP, but in your case it's possible that this fail.
If all is working well (all is green in status page and you only have the multipath message), check on host that it's not filtering MPTCP traffic. You can use tracebox from the router to check this or menu Network->MPTCP, MPTCP support check tab.

bob7gr, 2021-11-21:
Hi . thanks for the reply, as I said , the yellow warning message of "Multipath seems to be blocked on the connection" appears only when I use a vm created in my dedicated server as the VPS of  the openmptcprouter service and not when I used a standalone vps  with public IP from another company , as it is all green there. So , I guess it is the shorewall on my vm as it sees a private ip (10.0.2.15) instead of expected public ip? Any way of how to fix it? DNS is working on the vm as I can ping internet hosts (with names) without problems.

Ysurac, 2021-11-21:
As I said check with tracebox for Multipath. The problem can be on VPS/host side.
What is the public IP you get from OpenMPTCProuter ?

bob7gr, 2021-11-22:
Allright , I went to  "MPTCP support check" and did a trace and I got the following: (public IP and MACs are censored)

`1: 192.168.1.1 IP::DiffServicesCP IP::CheckSum 
2: {public_ip_appeared_as_2nd_hop} TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
3: {public_ip_appeared_as_3rd_hop} TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
4: {public_ip_appeared_as_4th_hop} TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
5: {public_ip_appeared_as_5th_hop} TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
6: {public_ip_appeared_as_6th_hop} TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
7: {public_ip_appeared_as_7th_hop}  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
8: {public_ip_appeared_as_8th_hop}  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
9: {public_ip_appeared_as_9th_hop}  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
10: {public_ip_appeared_as_10th_hop}  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
11: {public_ip_appeared_as_11th_hop}  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
12: {IP_OF_VM_VPS_HERE} TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = {SENSORED_MAC} , SourceMAC = {SENSORED_MAC}  , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 54 , Protocol = 0x6 , CheckSum = 0xb258 , SourceIP = {IP_OF_VM_VPS_HERE} , DestinationIP = {WAN_IP_OF_ROUTER_HERE} , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 34726 , SeqNumber = 2864211141 , AckNumber = 2046356127 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 64240 , CheckSum = 0x10af , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1452 , >`

Ysurac, 2021-11-22:
As you can see in the result there is "-TCPOptionMPTCPCapable" that mean MPTCP flag seem to be removed when reaching VPS side.

bob7gr, 2021-11-22:
So , that means that either the setup of the VPS (behind NAT) or the ISP of the datacenter giving the dedicated server is blocking MPTCP?

Ysurac, 2021-11-22:
This seems to be the case, at least if your VPS is using KVM and you rebooted to use mptcp kernel.

closed 2022-02-27 by github-actions[bot]