Ysurac/openmptcprouter #2186: QoS availability

sadahsan, 2021-11-23:
Does the system support end to end QoS support through shadowsocks and glorutun TCP or does it respect ToS markings? or any of those are dropped when passing between router and VPS?

Many Thanks

kencogit, 2021-11-23:
This interests me also if we have links that may vary or go up/down then there may be a need to prioritise one class of packet.

edaspb, 2021-11-24:
Same here. I'm looking for the option to make shadowsocks-libev transparent for ToS. For now all ToS marks are dropped. I found this related issue https://github.com/shadowsocks/shadowsocks-libev/issues/2591. Looks like shadowsocks-libev originally considered ToS trans-passing but there is a bug in implementation.

closed 2022-02-27 by github-actions[bot]