Ysurac/openmptcprouter #2189: Identifying bottleneck

Bestle, 2021-11-24:
I am trying to aggregate multiple high speed connections (combined 2.5Gbit) using MPTCP but seem to be hitting a bottleneck that is preventing me from achieving more than 1Gbit speeds.

I have a 2.5Gb connection from my PC to the OpenMPTCProuter machine which is verified as running at full speed and I am using a VPS with a 2Gb connection. Neither the VPS nor router max out on CPU / RAM during speed tests.

How would I go about verifying what the bottleneck is? Are there any settings I should change to achieve full performance?

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64

packlet, 2021-11-24:
"using a VPS with a 2Gb connection".  keep in mind the various connections to be aggregated have to go TO the VPS, and then the aggregated/bonded connection comes OUT of the VPS.  So, do you have 2gbps in BOTH directions simultaneously?  If they limit you to 2gbps total, then that would explain the ~1GB throughout limit you're seeing...

Bestle, 2021-11-24:
Good point, I was under the impression it was 2 up, 2 down, but could very well be 2 combined. Will double check and then try with a faster host.

Bestle, 2021-11-25:
1 down 2 up... useless. Before I close this issue, I don't suppose you know of any faster providers?

Bestle, 2021-11-25:
Switched to another VPS that tests at around 3 Gbps up and down but my bandwidth through OpenMPTCProuter is still capping out around 900 Mbps. Are there any configuration options I should be changing for MPTCP that will optimise for speed?

Bestle, 2021-12-02:
All sorted, pushing well over 1Gbps now. The CPU of my x86 box is struggling slightly at these speeds though.

https://www.speedtest.net/result/12399460267

packlet, 2021-12-02:
Nice!  Any info on what you needed to tweak, other than switching VPS, would be appreciated.  Also, for posterity, what are the specs on your x86 running the OpenMTCPRouter bonding?...  thanx

Bestle, 2021-12-03:
The main problem turned out to be a rubbish quality "CAT 6" cable that caused my PC to negotiate a 1 Gbps link rather than a 2.5Gbps one 9 times out of 10. I also disabled all shadowsocks encryption to give everything as much headroom as possible.

The CPU I am running is a i3-4010U.

closed 2022-03-08 by github-actions[bot]