Ysurac/openmptcprouter #2191: Aggregation stops for long sessions of rsync

Mazwak, 2021-11-26:
## Expected Behavior

Aggregation should work all the time, and recover from failed link

## Current Behavior

After a few hours of rsync for a big file  (> 20 GB), it uses only one connection.
If I disconnect this connection, rsync sees a closed connection and exits.

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: PulseHeberg
  - OpenMPTCProuter platform: RPi 4
  - Country: France
- - ADSL primary (Freebox cristal)
- - 4G enabled (Huawey B715s-23c with Free SIM)

Each device is connected through an USB-Ethernet, but I don’t think it matters.

Ysurac, 2021-11-26:
Do you have SQM enabled and download/upload speed configured ? I think the rsync may saturate connection and make a connection disconnect. You can check system log.

Mazwak, 2021-11-26:
SQM is disabled for both connections.

What should I look for in the logs ?

Ysurac, 2021-11-26:
So SQM should be enabled.
In the log you should look what can be related to a lost of connection/vpn/...

Mazwak, 2021-11-26:
I enabled SQM for ADSL and 4G.
I will report back tomorrow.

My web logs are flooded by :
 ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512

I can’t see more than a few minutes.

/var/log seems empty
Are there written logs somewhere else ?

Mazwak, 2021-11-26:
I think it happened.

ADSL :
[image]

4G :
[image]

The log from web is already filled with :
Fri Nov 26 17:56:44 2021 kern.err kernel: [4230581.033758] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Fri Nov 26 17:56:47 2021 kern.err kernel: [4230584.043056] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512

and it begins at 17:25 :(

Is the system log written somewhere ?

Ysurac, 2021-11-26:
You updated from a previous version ? there is some bugs in wifi RPI4 drivers, and using some settings seems to solve log problem.
To remove this warning, you need in Network->Wireless, to set a country, and may also need to set a mode, band and channel in operating frequency.

Mazwak, 2021-11-26:
I don’t use wifi anyway. Is there a way to disable it totally ?

Ysurac, 2021-11-26:
Maybe by removing the module using ```rmmod brcmfmac```

Mazwak, 2021-11-27:
Just happened. Here is the « system log » :

```
Sat Nov 27 16:32:56 2021 daemon.err /usr/bin/ss-redir[7371]: remote recv: Operation timed out
Sat Nov 27 16:40:03 2021 daemon.err /usr/bin/ss-redir[7371]: remote recv: Operation timed out
Sat Nov 27 16:40:05 2021 daemon.err /usr/bin/ss-redir[7374]: remote recv: Operation timed out
Sat Nov 27 16:40:07 2021 user.notice post-tracking-post-tracking: Check API configuration...
Sat Nov 27 16:40:18 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Sat Nov 27 16:40:26 2021 daemon.err /usr/bin/ss-redir[7371]: remote recv: Operation timed out
Sat Nov 27 16:40:32 2021 daemon.err /usr/bin/ss-redir[7371]: remote recv: Operation timed out
Sat Nov 27 16:40:57 2021 daemon.err /usr/bin/ss-redir[7371]: remote recv: Operation timed out
Sat Nov 27 16:41:41 2021 daemon.err omr-tracker[32357]: Segmentation fault
Sat Nov 27 16:41:43 2021 daemon.err omr-tracker[32358]: Segmentation fault
Sat Nov 27 16:41:46 2021 daemon.err omr-tracker[32357]: Segmentation fault
```

I have one rsync, and one game update which are ADSL only after that.

I noticed it because I was disconnected from the online game I was playing.

I stopped and started game update, and it uses aggregation again.

Mazwak, 2021-11-27:
Surprisingly, the rsync recovered. I don’t know when exactly, but between 17:42 and 19:05.

Logs are 1h off btw.

Mazwak, 2021-11-28:
Happened during the night.
Download started around 1 am.
It never recovered. It uses ADSL only.

Sun Nov 28 02:00:00 2021 cron.err crond[4056]: USER root pid 16476 cmd /etc/init.d/vnstat_backup backup
Sun Nov 28 02:00:00 2021 cron.err crond[4056]: USER root pid 16477 cmd /etc/init.d/omr-bypass bypass_asn
Sun Nov 28 02:35:16 2021 daemon.err /usr/bin/ss-redir[12876]: remote recv: Operation timed out
Sun Nov 28 02:41:43 2021 user.notice post-tracking-post-tracking: Check API configuration...
Sun Nov 28 02:41:53 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Sun Nov 28 02:44:52 2021 daemon.err /usr/bin/ss-redir[12875]: remote recv: Operation timed out
Sun Nov 28 02:48:35 2021 daemon.err /usr/bin/ss-redir[12878]: remote recv: Operation timed out
Sun Nov 28 03:56:27 2021 daemon.err /usr/bin/ss-redir[12876]: remote recv: Operation timed out
Sun Nov 28 04:00:00 2021 cron.err crond[4056]: USER root pid 13436 cmd /etc/init.d/vnstat_backup backup
Sun Nov 28 04:03:02 2021 daemon.err /usr/bin/ss-redir[12878]: remote recv: Connection reset by peer
Sun Nov 28 04:40:28 2021 user.notice post-tracking-post-tracking: Check API configuration...
Sun Nov 28 04:40:38 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Sun Nov 28 04:43:16 2021 daemon.err /usr/bin/ss-redir[12875]: remote recv: Operation timed out
Sun Nov 28 04:51:50 2021 daemon.info glorytun[12895]: STOPPED tun0
Sun Nov 28 04:51:50 2021 daemon.err glorytun[12895]: 89.234.182.63.65001: connected
Sun Nov 28 04:51:50 2021 daemon.info glorytun[12895]: STARTED tun0
Sun Nov 28 04:56:41 2021 daemon.err /usr/bin/ss-redir[12875]: remote recv: Operation timed out
Sun Nov 28 05:03:57 2021 daemon.info dnsmasq[15342]: read /etc/hosts - 4 addresses
Sun Nov 28 05:03:57 2021 daemon.info dnsmasq[15342]: read /tmp/hosts/odhcpd - 4 addresses
Sun Nov 28 05:03:57 2021 daemon.info dnsmasq[15342]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Sun Nov 28 05:03:57 2021 daemon.info dnsmasq-dhcp[15342]: read /etc/ethers - 0 addresses
Sun Nov 28 06:00:00 2021 cron.err crond[4056]: USER root pid 12099 cmd /etc/init.d/vnstat_backup backup
Sun Nov 28 06:39:11 2021 user.notice post-tracking-post-tracking: Check API configuration...
Sun Nov 28 06:39:20 2021 user.notice post-tracking-post-tracking: Check API configuration... Done

[image]

Mazwak, 2021-11-28:
I noticed that when rsync proceeds to the next file, it might recover. Not always, but it does sometimes.

closed 2022-03-04 by github-actions[bot]