Ysurac/openmptcprouter #2194: Custom DNS Over HTTPS

matfil92, 2021-11-29:
Hi,
how can I setup my own custom DNS Over HTTPS ?

I've ADGuard Remote Server, custom one not the public version, and I want understand how I can setup my custom DoH without insert the DNS IP in Network --> DCHP and DNS but setup custom DoH in Services --> DNS Over HTTPS Proxy Settings

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS provider: Aruba
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-11-29:
You can use unbound, used by default with DNS root servers, in Services->recursive DNS, "Zones" tab.

matfil92, 2021-11-29:
When setup Recursive DNS tab I can remove the DNS from DHCP and DNS under Network menu?

Ysurac, 2021-12-01:
No. DNS is redirected from dnsmasq (configured in DHCP and DNS) to unbound (configured in recursive DNS).