Ysurac/openmptcprouter #2195: "No data" on status page (not concerning USB modems) on Pi4

Rookie805, 2021-11-29:
## Expected Behavior

After setting OMR up with WAN-1 interface as eth0.2 (VLAN with DHCP client) and WAN-2 as eth1 (Huawei 3372h in HiLink mode with DHCP enabled) I see status page with all modules green, OMR works using both WAN interfaces, succesfully getting IP's from both DHCPs.

## Current Behavior

 After powering device on the next day I get "No data" on status page as soon as I connect WAN-1 device and OMR gets IP address by DHCP. Disconnecting WAN-1 and connecting WAN-2 gives same result.
Didn't find anything about it in logs. 

Despite blank status page OMR seems working, I started writing this issue, but www.speedtest.net never gets finished, ending with error.

## Possible Solution

Have no idea.
## Steps to Reproduce the Problem

  1. Power OMR up, connect to switch
  2. Attach WAN-1 to switch
  3. Get "No data" error.

## Context (Environment)
System log:
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[9] (Allow-All-LAN-to-VPN) does not specify a protocol, assuming TCP+UDP
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[12] (Allow-Lan-to-Wan) does not specify a protocol, assuming TCP+UDP
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface ip6tnl0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ip6tnl0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface sit0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=sit0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface gre0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=gre0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface gretap0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=gretap0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface erspan0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=erspan0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface ip6gre0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ip6gre0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface bond0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=bond0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface teql0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=teql0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice Shadowsocks: Rules UP
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: add network interface wlan0: event processed
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: event reported: action=add, name=wlan0, subsystem=net
Sat Nov 27 21:03:23 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov 27 21:03:23 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'glorytun'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[9] (Allow-All-LAN-to-VPN) does not specify a protocol, assuming TCP+UDP
Sat Nov 27 21:03:23 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[12] (Allow-Lan-to-Wan) does not specify a protocol, assuming TCP+UDP
Sat Nov 27 21:03:24 2021 daemon.notice wpa_supplicant[3420]: Successfully initialized wpa_supplicant
Sat Nov 27 21:03:24 2021 user.notice : Added device handler type: 8021ad
Sat Nov 27 21:03:24 2021 user.notice : Added device handler type: 8021q
Sat Nov 27 21:03:24 2021 user.notice : Added device handler type: macvlan
Sat Nov 27 21:03:24 2021 user.notice : Added device handler type: veth
Sat Nov 27 21:03:24 2021 user.notice : Added device handler type: bridge
[... 1340 lines cut ...]
[   12.518816] bcmgenet fd580000.ethernet eth0: Link is Down
[   13.056385] usb 1-1.1: new high-speed USB device number 4 using xhci_hcd
[   13.165421] usb 1-1.1: New USB device found, idVendor=12d1, idProduct=14dc, bcdDevice= 1.02
[   13.173842] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   13.181164] usb 1-1.1: Product: HUAWEI_MOBILE
[   13.185524] usb 1-1.1: Manufacturer: HUAWEI_MOBILE
[   13.191069] usb 1-1.1: no of_node; not parsing pinctrl DT
[   13.269681] cdc_ether 1-1.1:1.0: no of_node; not parsing pinctrl DT
[   13.306889] cdc_ether 1-1.1:1.0 eth1: register 'cdc_ether' at usb-0000:01:00.0-1.1, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[   13.320825] uas 1-1.1:1.2: no of_node; not parsing pinctrl DT
[   13.327113] usb-storage 1-1.1:1.2: no of_node; not parsing pinctrl DT
[   13.333672] usb-storage 1-1.1:1.2: USB Mass Storage device detected
[   13.341744] scsi host0: usb-storage 1-1.1:1.2
[   14.365945] scsi 0:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
[   14.374268] sd 0:0:0:0: no of_node; not parsing pinctrl DT
[   14.380556] sd 0:0:0:0: Power-on or device reset occurred
[   14.397393] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   16.604421] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026

  - OpenMPTCProuter platform: Rpi4-8GB

Ysurac, 2021-11-29:
Do you know where "10.37.155.104/29" is coming from ?

Rookie805, 2021-11-29:
Yes, it is WAN IP issued by LTE operator. The LTE modem is in bridge mode.

lotherk, 2021-12-01:
I have the feeling the problem comes from a wrong time/date set on the device. I had the same behaviour on my rpi3 and noticed that the time/date was off. I set time manually and had the data back. I installed and enabled ntpd and since then it's working.

closed 2022-03-07 by github-actions[bot]

Candyffm, 2022-05-16:
> I have the feeling the problem comes from a wrong time/date set on the device. I had the same behaviour on my rpi3 and noticed that the time/date was off. I set time manually and had the data back. I installed and enabled ntpd and since then it's working.

this error drove me nuts. Thanks man!

darkmaniac7, 2022-06-06:
Wanted to also say thanks, I'd have never thought that was the issue!

Drove me crazy for hours after changing hardware.