Ysurac/openmptcprouter #2197: Rpi4 -----> Rpi3 restore backup on new raspberry 3 changes mac address to the old one.

highfreq, 2021-11-30:
## Current Behavior
Restoring a backup on a different Raspberry uses mac address of old raspberry

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1026
  - OpenMPTCProuter VPS provider: Aruba - Italy
  - OpenMPTCProuter platform:  RPi4 (old) RPi3(new)

I used to have an Rpi4 as router that i needed for another project. I made a backup of it than installed OMR on an Rpi3 and restored the backup done on Rpi4. After the restore the mac address of the new Rpi3 is wrong and still has the one from the old raspberry. 

Is there any way to tell the system to restore backup but not mac addresses or chenge addresses to correct ones after restore?

Thanks

Ysurac, 2021-11-30:
You need to modify the backup. It's a compressed file, you need to remove lines that begin with "option macaddr" in ```etc/config/network``` inside.

highfreq, 2021-11-30:
Thank you, i cleaned all mac addresses, raspberry, wifi, and the 2 modems and restored the backup. It took all the correct addresses of the board installed.
It worked great!!!!
Thank you.

closed 2022-03-06 by github-actions[bot]