Ysurac/openmptcprouter #2198: No internet for some clients

ThibautBascoulergue, 2021-12-01:
- OpenMPTCProuter version: 58.5
  - OpenMPTCProuter VPS version: 0.1.26.5.4.100
  - OpenMPTCProuter VPS provider: Firstheberg
  - OpenMPTCProuter platform: vmware ova

Hello,

Since yesterday I have encountered a problem that I cannot resolve. I have two 4G connections and since yesterday I have one to fall. Normally it does not impact internet access except that since yesterday, some Wi-Fi devices do not access certain sites. Example my phone, once connected to Wifi, I do not access youtube or Facebook, on the other hand via the Chrome browser everything works. I have looked at the logs and the only thing that I do not find normal are these lines.

Wed Dec  1 10:49:58 2021 daemon.info dnsmasq[14953]: 27 192.168.1.136/61709 validation result is INSECURE

Thx

hairy-dog, 2021-12-01:
Is the 4G that stopped set as Master?

ThibautBascoulergue, 2021-12-01:
Hello, no and i have réinstalled vps and router, same problem. I think the problem is the dns.

Ysurac, 2021-12-01:
This seems to be a problem with DNSSEC.
Check that router and clients time are accurate. If all seems to be ok, you can try to disable DNSSEC in Services->Recursive DNS

closed 2022-03-07 by github-actions[bot]