Ysurac/openmptcprouter #2200: support for rb750gr3 ( mikrotik )

epictz06, 2021-12-02:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Hello , is there anyway to flash OMR on this router? i need only one shadowsocks package and mptcp enabled , i want to ask is it possible to flash OMR on this router?

Ysurac, 2021-12-06:
It have a slow CPU and a small flash.
You can try to compile it yourself using https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

closed 2022-03-11 by github-actions[bot]

thelastdreamer, 2022-10-16:
Hello, right now mikrotik support even containers or even direct openwrt with many models supported.

Is there any image which i can use to install?
It would be great if you create something like this for mikrotik since they are value for money.