Ysurac/openmptcprouter #2202: Raspberry CM4

highfreq, 2021-12-04:
Is OMR compatible with CM4?
Would like to use this:

https://raspberrypi.dk/en/product/iot-router-cm4-carrier-board-mini/?currency=EUR&gclid=CjwKCAiAwKyNBhBfEiwA_mrUMvrg6qncFkC4H2jH0LV1P9t4K36FO85Qz7z5X1oVDTlqikIXBpOIlhoCRQoQAvD_BwE

Ysurac, 2021-12-06:
This may be supported in next release.

highfreq, 2021-12-06:
When will new release be out?

highfreq, 2021-12-08:
It would be great if it could be supported, it is small, powerful enough and with 2 ethernet ports.

highfreq, 2022-01-27:
@Ysurac , have any news on CM4 support? Would like to use it on a new project.

Ysurac, 2022-01-27:
Snapshot should support CM4

highfreq, 2022-01-27:
Is it stable enough to be used?

Can i use emmc on board?

highfreq, 2022-01-27:
Found the snapshots but a bit lost, would you mind give me a link for the correct image to be used?

thanks,

Corrado

Ysurac, 2022-01-27:
I've a daily job, I answer on free time only...
Images are here: https://download.openmptcprouter.com/develop/5.4/rpi4/targets/bcm27xx/bcm2711/
Should be quite stable, OMR VPS snapshot may be needed too: `wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh`
It's not tested on CM4.

highfreq, 2022-02-03:
@Ysurac VPS snapshot is the regular debian 10 vps, looks like so from the link you provided.

Ysurac, 2022-02-03:
no, note the "server-test" in the link.

highfreq, 2022-02-03:
Ooooops sorry, missed it, thanks

kamileq, 2022-03-30:
So using some development version not the newest beta release to figure out it on RP CM4? Will it work with eMMC solution not microSD card?

Ysurac, 2022-03-30:
Latest beta should work too.

closed 2022-07-03 by github-actions[bot]