Ysurac/openmptcprouter #2204: streaming

ascomputer, 2021-12-06:
hello everyone, I use 3 connections in 4g and during a streaming upload if a 4g connection drops and then comes back active again it does not work anymore I have to disconnect the streaming and then reactivate ... how can I solve? Thank you

Ysurac, 2021-12-06:
Please use the issue template.

ascomputer, 2021-12-07:
forgive me but I don't understand your request

Ysurac, 2021-12-07:
When an issue is open, there is a template: https://github.com/Ysurac/openmptcprouter/issues/new?assignees=&labels=help+wanted&template=help.md
With expected behavior, current behavior and specification. This help to know what is the issue (and can help to reproduce it) and on what version.

ascomputer, 2021-12-07:
Use this model, otherwise the problem may be closed automatically ->
## Expected behavior
<! --- In English please. ->
<! --- Tell us what should happen ->
I use 3 4g connections to stream on youtube while streaming a 4g connection went down then came back but didn't work anymore. to make it work I had to unplug the stream and restart it.
on the net --- mptcp is set fullmesh roundrobin - mctcpdesync
## Actual behavior
<! --- In English please. ->
<! --- Tell us what happens instead of the expected behavior ->
<! --- Did you check the register? In Status-> Registry ->

## Specifications

   - OpenMPTCProuter version: <! --- (The latest is not a version) -> V0.57.3
   - OpenMPTCProuter VPS version: <! --- (The latest is not a version) -> latest version
   - OpenMPTCProuter VPS Provider:
   - OpenMPTCProuter platform: <! --- (RPI2 / RPI3 / x86 / x86_64) -> x86_64

<! --- (please do not attach text files) ->

closed 2022-03-13 by github-actions[bot]