Ysurac/openmptcprouter #2205: multitcp blocked on network

benw8080, 2021-12-06:
I'm wondering if there is a possible work around, when this is error is given the network connection becomes unusable in OMR.

"Multitcp is blocked on this connection" - this can present when using a "corporate or managed" network or 4G mobile network.

I am using a rasPi 4 with OMR v0.58.5.
I am using independant USB - ethernet NIC's for all WAN's on the pi and the eth0 for the LAN connection.

When connecting to a corporate networkOMR gives an error saying "multitcp is blocked" and zero traffic is able to be routed by OMR across this connection. The connection's are 400mb+ bandwidth. When used into a non OMR router the connection routes to internet etc normally.

I have also received this error when connecting to a 4G LTE carrier, this can be intermittent.

Please let me know what further information I can offer to assit with resolving this, or steps we can take to work around the error msg.

Thanks for all the effort with OMR, its a great platform.

homicida2000, 2021-12-09:
You can tunnel the MPTCP connection through a VPN on the interface.
Use Glorytun-UDP or Wireguard

Ysurac, 2021-12-10:
In the wizard you can check "MPTCP over VPN" that is made when something block MPTCP.
To check where it's blocked, you can use Network->MPTCP, "MPTCP Support Check" tab.

benw8080, 2021-12-14:
thanks ysurac and Homicida2000

closed 2022-03-20 by github-actions[bot]