Ysurac/openmptcprouter #2206: SER2NET

highfreq, 2021-12-08:
@Ysurac Do you think ser2net can be installed without breaking everything?

Ysurac, 2021-12-10:
Should not break anything if the TCP port you set for ser2net is not already used.

highfreq, 2021-12-10:
Ok great i'll try. Do you think i can use luci interface for ser2net too?

Ysurac, 2021-12-10:
yes, no problem.

highfreq, 2021-12-10:
I get a tons of "no valid architectures" during installation and it aborts it. I am running 0.58 on a raspberry Pi3

Ysurac, 2021-12-10:
You need to follow https://github.com/Ysurac/openmptcprouter/issues/2099 to fix package list update

highfreq, 2021-12-10:
Tried all solutions in #2099 none of them worked:
* opkg_install_cmd: Cannot install package libustream-openssl20201210.

than tried:

cd /etc/ssl/certs
rm DST_Root_CA_X3.crt ca-certificates.crt
cat *.crt > ca-certificates.crt

no errors but still can't download packages.

than tried:
opkg install wget-ssl
* opkg_install_cmd: Cannot install package wget-ssl.

highfreq, 2021-12-11:
@Ysurac anything else i can try?

highfreq, 2021-12-11:
Is there a way to install packages other than what you linked wich i already tried and didn't work?

highfreq, 2021-12-17:
@Ysurac All solutions provided fail to install with description posted, is there any way to solve please?