Ysurac/openmptcprouter #2207: Monitor and save WAN bandwidth

eothein, 2021-12-08:
Using the Status->Real Time Graphs I'm able to monitor my WAN connections, however, I would like to see the raw data (i.e. using a CSV file) of a certain time period. 

Is there an easy way to do this?

Ysurac, 2021-12-10:
For this graph, "luci-bwc" is used. There is also vnstat that is installed and can be easier to use.

jjak0b, 2021-12-13:
> Using the Status->Real Time Graphs I'm able to monitor my WAN connections, however, I would like to see the raw data (i.e. using a CSV file) of a certain time period.
> 
> Is there an easy way to do this?

I usually use "nlbwmon".
I manually added some protocols names to port mapping but it helped me to investigate on bandwidth usages.
It could be improved a lot but does its work.

closed 2022-03-18 by github-actions[bot]