Ysurac/openmptcprouter #2210: Support to Ysurac

highfreq, 2021-12-10:
Just want to say a public thank you to Ysurac, to demonstrate our gratitude we just started supporting him on patreon. I think all of us that use his software should show him some real support with some donation or subscription, based on the possibilities each of us has.

I think it was deserved,

thank you

wlatic, 2021-12-13:
One thing that could help massively is having a small recurring donation able to be easily setup.

Being able to easily donate a couple of dollars a move, for most people, wouldn't be an issue and for Ysurac it could really help to get some funds together for both personal and to expand the project if required.

Bestle, 2021-12-14:
> One thing that could help massively is having a small recurring donation able to be easily setup.
>
> Being able to easily donate a couple of dollars a move, for most people, wouldn't be an issue and [...]

You can do that here: https://www.patreon.com/ycarus

GTiDon, 2022-02-22:
Is Ysurac still around? I haven't seen any responses for a long time now

highfreq, 2022-02-22:
Sure he is, otherwise we wouldn't get any update :) 
He posted in several threads lately.

closed 2022-05-29 by github-actions[bot]