Ysurac/openmptcprouter #2214: wireguard tunnel slow speed, mtu issue?

ngarafol, 2021-12-13:
Hi, my OMR installation is pretty much working fine. 

Issue I am facing is wireguard slow tunnel speed. 

My wireguard setup is following:

client A: regular xdsl line, 50/10mbit avg
client B: behind OMR (using OMR as gateway), 3 links, combined speed 160/80mbit avg
wireguard server: vps in france (not the one running OMR server part) speedtest over 1/1gbit to all related client A/B isp/destinations.

When copying files from client B to client A I would expect wirespeed, so in theory 50mbit since that is client A downlink. But most I get is 10-15mbit in best case. I see 2 links being used on client B during transfer, so there is capacity left. Also capacity free on client B side since any other download gets me wirespeed. Server has plenty of bandwidth. I could not identify any hw bottleneck on any part of wireguard setup. In OMR all three connections show MTU of 1500. Wireguard uses default of 1420. 

I am thinking, should I try with something lower for wireguard mtu, like 1340 maybe? 1340 being 1500 -80 -80 to be on the safe side, since Wireguard needs 80 and  I believe OMR still needs some.

Ysurac, 2021-12-16:
Wireguard use UDP, you can try another VPN in the wizard, advanced settings checkbox. By default glorytun TCP is used, sometimes another VPN work better.

ngarafol, 2021-12-16:
Thank you for suggestion. Currently I enabled v2ray, and it looks like wireguard speed improved.

closed 2021-12-16 by ngarafol