Ysurac/openmptcprouter #2215: OMR-Bypass with Ipv6

TKR-US, 2021-12-14:
## Expected Behavior

OMR-Bypass won't work with ipv6, based on AS Nimber or IP or subnet

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem
On VPS, i've 2 Ipv6 Subnet.
Configure one IPV6 Subnet on EULA Openmptcprouteur.
CLients have one ipV4 and one ipv6 of the public subnet.

Added AS Number, and IPS V6 on OMR-Bypass

Tested on google.com / fr
WIth Ipv4 only, https://www.google.com/sorry/index give ipV4 of box
With Ipv6, https://www.google.com/sorry/index give ipv6 of my subnet, not ipv6 of my box (or redirect on ipv4 box)

It is a bug ?

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: Debian 10
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: RPI4
  - Country:

Ysurac, 2021-12-16:
Did you set IPv6 of the interface used for ByPass ?

TKR-US, 2021-12-16:
Hello,
i've  modify my configuration, but same issue for ipv6;

First configuration :
- Hetzner VPS with one ipv6 IP for VPS
- a second ipv6 subnet, not configuration of this subnet on vps
- I Put this New subnet on Préfixe ULA IPv6

Second configuration (actual configuration)
- vtunnel /48 of he.net, On the VPS
- on Préfixe ULA IPv6, a /64 of the /48HE.

Example of test with ip.lafibre.info:
 Ipv4: 45.85.134.187
Ipv6: 2a05:46c0:100:1007::3
AS: 200780

Added all in OMR, apply... waiting.....

results: 
Ipv4: the vdsl box
ipv6: IpV6 of HE...

Probably bug is Orange boxes...
After reboot the box, in network: Le préfixe IPv6 est : N/A

So, the bug is Orange, not openmptcprouter.. Right?

behrndt, 2021-12-20:
Hello,
I think I have got the same issue in a similar setup:

I have two WAN connections:
- Cable 1000/50 currently IPv4 only (master)
- DSL 250/45 IPv6 + IPv4 via DSLite (MPTCP via VPN with Wireguard enybled on IPv4-DHCP-Interface only, second Interface configuration with DHCPv6 for same physical interface has MPTCP disabled)

VPS at Hetzner.
LAN with IPv4 only.

Until now all was running fine with IPv4 only including OMR Bypass for several domains, IPs and protocols through both of the direct WAN connections.

I now wanted to enable IPv6 on my network.
Hetzner VPS has one IPv6/64 used only by the VPS and a second IPv6/64 that is assigned to my OMR-Router.
I have enabled IPv6 on my LAN interface with a static IP and the second /64-Prefix
All of that worked fine so far, IPv6 is working on the local network and to the internet through VPS assigned IP.

But OMR-Bypass ist completely not working for IPv6 (tested with IP, domain and protocol) while bypass is still working for IPv4-connections. Clients are using OMR as the only DNS-server (via IPv4 and IPv6).
After some research I tracked it down to be some issue with the script /etc/init.d/mptcp
The IPsets for OMR-Bypass are correctly built in /etc/init.d/omr-bypass also for IPv6.
But the routing rules and table lookups for the corresponding fwmark are not built in /etc/init.d/mptcp.
The script simply does not seem find the IPv6-address, network, gateway and netmask for the DHCPv6-interface of my DSL WAN-connection so the routing rules are not created at all.

I did no yet manage to adjust the script in a way that it can detect all the IPv6-parameters correctly.
Also, after the script is running the first time it seems completely destroy the DHCPv6-lease on the DSL wan interface because after some minutes the IPv6-connection on this interface gives "permission denied" erros in the log and no IPv6 pings are possible anymore through this interface and the previously assigned IPv6-adresses (link and global scope) are lost.
Then for sure, the bypass cannot work though the interface.

@Ysurac is there maybe a newer development version existing of the script parts for IPv6? (see my versions below).

OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
OpenMPTCProuter VPS provider: Hetzner
OpenMPTCProuter platform: x86 (Intel(R) Core(TM) i7-4770S CPU @ 3.10GHz) Proxmox-Environment
Country: Germany