Ysurac/openmptcprouter #2216: 5 connections, but only 2 aggregates

Pokpok-sbo, 2021-12-15:
Hello everyone !
## Current behavior
My bandwidth has become very low compared to my connections.
When I look at the statuses, everything is apparently perfect.

[OpenMPTCProuter - État - LuCI.pdf](https://github.com/Ysurac/openmptcprouter/files/7717872/OpenMPTCProuter.-.Etat.-.LuCI.pdf)

On the graphics, it appears that there are only two connections working.
And in the FullmeshMPTCP tab, visibly there are only 2 aggregated connections.

[image]

Do you have any ideas for OPENMPTCP to use all 5 connections?

THANK YOU!

## Characteristics

  - OpenMPTCProuter version: openmptcprouter v0.58rc2 r0 + 16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1025-test
  - OpenMPTCProuter VPS provider:
  - OVH OpenMPTCProuter: Model ： Dell Inc. Precision WorkStation 490
Architecture ： Intel (R) Xeon (R) CPU 5160 @ 3.00GHz

Ysurac, 2021-12-16:
Can you put the capture of the status page as image instead of PDF ? thanks.

Pokpok-sbo, 2021-12-17:
[image]

Ysurac, 2021-12-17:
I would need, in System->OpenMPTCProuter, "Show all settings" tab, the part that begin with "network." and in Status->routes, the "Active IPv4-Routes" part.

Pokpok-sbo, 2021-12-17:
Show all settings:

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].dnssec='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs0,omr_dscp6-cs0,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/aircall.io/omr_dscp-ef,omr_dscp6-ef' '/amazonaws.com/omr_dscp-ef,omr_dscp6-ef' '/twilio.com/omr_dscp-ef,omr_dscp6-ef' '/chunderwgll.twilio.com/omr_dscp-ef,omr_dscp6-ef' '/chunderw-gll.twilio.com/omr_dscp-ef,omr_dscp6-ef' '/chunderw-vpc-gll.twilio.com/omr_dscp-ef,omr_dscp6-ef' '/eventgw.twilio.com/omr_dscp-ef,omr_dscp6-ef' '/matrix.twilio.com/omr_dscp-ef,omr_dscp6-ef' '/media.twiliocdn.com/omr_dscp-ef,omr_dscp6-ef' '/static.intercomcdn.com/omr_dscp-ef,omr_dscp6-ef' '/ws.pusherapp.com/omr_dscp-ef,omr_dscp6-ef' '/pusher.com/omr_dscp-ef,omr_dscp6-ef' '/stun.zoiper.com/omr_dscp-ef,omr_dscp6-ef' '/zoiper.com/omr_dscp-ef,omr_dscp6-ef' '/sylho.allocloud.com/omr_dscp-ef,omr_dscp6-ef' '/allocloud.com/omr_dscp-ef,omr_dscp6-ef' '/app.ringover.com/omr_dscp-ef,omr_dscp6-ef' '/sip.ringover.com/omr_dscp-ef,omr_dscp6-ef' '/ringover.net/omr_dscp-ef,omr_dscp6-ef' '/www.banky-foibe.mg/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/ringover.com/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/meet.google.com/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/mcbmadagascar.com/omr_dst_bypass_eth1,omr6_dst_bypass_eth1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.Vlan5=dhcp
dhcp.Vlan5.interface='Vlan5'
dhcp.Vlan5.start='100'
dhcp.Vlan5.limit='150'
dhcp.Vlan5.leasetime='12h'
dhcp.Vlan10=dhcp
dhcp.Vlan10.interface='Vlan10'
dhcp.Vlan10.start='100'
dhcp.Vlan10.limit='150'
dhcp.Vlan10.leasetime='12h'
dhcp.Vlan11=dhcp
dhcp.Vlan11.interface='Vlan11'
dhcp.Vlan11.start='100'
dhcp.Vlan11.limit='150'
dhcp.Vlan11.leasetime='12h'
dhcp.Vlan12=dhcp
dhcp.Vlan12.interface='Vlan12'
dhcp.Vlan12.start='100'
dhcp.Vlan12.limit='150'
dhcp.Vlan12.leasetime='12h'
dhcp.Vlan13=dhcp
dhcp.Vlan13.interface='Vlan13'
dhcp.Vlan13.start='100'
dhcp.Vlan13.limit='150'
dhcp.Vlan13.leasetime='12h'
dhcp.Vlan14=dhcp
dhcp.Vlan14.interface='Vlan14'
dhcp.Vlan14.start='100'
dhcp.Vlan14.limit='150'
dhcp.Vlan14.leasetime='12h'
dhcp.Vlan15=dhcp
dhcp.Vlan15.interface='Vlan15'
[... 2922 lines cut ...]
v2ray.omrtest.listen='127.0.0.1'
v2ray.omrtest.s_socks_auth='noauth'
v2ray.omrtest.s_socks_udp='1'
v2ray.omrtest.s_socks_ip='127.0.0.1'
v2ray.omrtest.s_socks_userlevel='0'
v2ray.main_reverse=reverse
v2ray.main_reverse.enabled='1'
v2ray.main_reverse.bridges='omrbridge|omr.lan'
v2ray.omrexit=outbound
v2ray.omrexit.protocol='freedom'
v2ray.omrexit.tag='out'
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth2' 'eth3' 'eth4' 'eth1' 'eth0.5' 'eth0.10' 'eth0.11' 'eth0.12' 'eth0.13' 'eth0.14' 'eth0.15' 'eth0.21' 'eth0.22' 'eth0.23' 'eth0.24' 'eth0.32' 'eth0.33' 'eth0.34' 'eth5'

Pokpok-sbo, 2021-12-17:
[image]

Pokpok-sbo, 2021-12-17:
[image]

mstrolo, 2021-12-31:
did you manage to fix it?

closed 2022-04-05 by github-actions[bot]