Ysurac/openmptcprouter #2221: Problem with OpenMPTCPRouter as vm in VirtualBox (windows host)

bob7gr, 2021-12-18:
Hello , I am not sure if this is an issue belonging only to OMR (OpenMptcpRouter) or OpenWRT in general.  I use OMR as a vm in a windows headless machine. I have attached a USB-Etherned adapter in order to act as the WAN interface of the router but sometimes , in kernel log of OMR , I see a bunch of errors , like the following picture: [https://imgur.com/a/j0EmvMv](url) Any ideas of what is wrong here?

Ysurac, 2021-12-20:
Please use the template.
What is the adapter model ?

bob7gr, 2021-12-21:
Hi, it is a TP-Link USB-gigabit ethernet adapter, I went ahead and disabled USB completely on the OMR vm in order to get internet directly from host interfaces (Bridge with adapter) and that looked like it fixed it, no errors yet , because usb network adapters are now directly connected to host machine only.

closed 2022-03-26 by github-actions[bot]