Ysurac/openmptcprouter #2226: Low Speed at Client when using UDP based VPN (Wireguard/OpenVPN) on it with Glorytun UDP

sebaro11, 2021-12-23:
Adequate speed when using a UDP based VPN on a Client behind the OpenMPTCPRouter with Glorytun UDP.

Currently I'm using Shadowsocks and Glorytun UDP als Proxy and VPN for OpenMPTCPRouter. Speed is very decent, until the point I connect a client to a Wireguard or OpenVPN Server which uses UDP as protocol. Tested with my private WireGuard Server and the VPN provider AirVPN.
My Speed with UDP connections over the VPN provider is capped at roundabout  16 Mbit/s.

Changing the VPN for OpenMPTCPRouter to Glorytun TCP, I can achieve much more higher speeds, nearly whats possible without VPN. But the latency is (of course through the behavior of TCP) higher to the Glorytun UDP connection.

So my question is, if this is an common/expected behavior for Glorytun UDP and if there is a way to achieve the same speeds with it like Glorytun TCP.

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: Hetzner Cloud (Nürnberg)
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2

Ysurac, 2021-12-23:
You can try V2Ray as proxy as TCP and UDP will use it.

sebaro11, 2021-12-23:
Thank you for the answer!
This afternoon I tried V2Ray with Glorytun UDP, but I think the Pi 4 does not has enough performance for it.
The speed raised (TCP and UDP VPN tested) to ~50 Mbit/s down/up.
Sticking currently with Shadowsocks and Glorytun TCP as it gives the best results.
So it seems that there is no "quick fix" possible. 😅

closed 2021-12-29 by sebaro11